FPGAwars/apio #286: [board request] APIO support for the iceWerx Board

ertgtct, 2022-02-15:
It would be graet to have support for this low cost FPGA Board:
https://www.robotshop.com/en/devantech-icewerx-ice40-hx8k-fpga.html
It should be feasable, because it uses the ice40hx8k and is programmed with the iceFun programmer.
Thanks!

Obijuan, 2022-02-18:
Yes, it is feasable. I did not know this board. I will buy one and add support to it (I can't tell you how much time it will take as I have very little time). But I will add to my todo-list

Thanks for the suggestion

Obijuan, 2022-04-28:
I have just received my iceWerx board. Support in apio and icestudio are coming 😀️

Obijuan, 2022-04-28:
The board is already supported. It was easy to add to apio because it is very similar to the iceFun board, that was already supported.
I will close this issue once the board is fully suported in Icestudio

Obijuan, 2022-05-02:
Support in Icestudio is ready!  Icestudio 0.9.2w202205020905 or higher

closed 2022-05-02 by Obijuan