FPGAwars/apio #287: system.vpi could not be found

classic-sasquatch-behavior, 2022-02-21:
howdy! trying to run a testbench and I'm being denied usage of some functions ($dumpfile etc.). I believe this is because of another error that it's throwing:
error: Failed to open 'C:\Users\USER~1\APIO~1\packages\TOOLS-~2\lib\ivl\system.vpi' because:
: The specified module could not be found.

system.vpi is indeed in this location. Why then can't apio find it?
many thanks.

classic-sasquatch-behavior, 2022-02-21:
to be more descriptive, it's in .apio\packages\tools-oss-cad-suite\lib\ivl. I'm on the most recent version of apio.

classic-sasquatch-behavior, 2022-02-21:
I just had to add the mentioned directory to path... no problem here

0xECEAD, 2022-04-16:
Hi, I have the same problem (on windows, with 3 different systems).

After some digging around, it turns out that system.vpi is actually a dll-file. And the problem is that it cannot load some of its dependencies (_libbz2-1.dll, libgcc_s_seh-1.dll, libwinpthread-1.dll, libstdc++6.dll and zlib1.dll_). They are not included in the package (tools-oss-cad-suite-windows-0.0.7). The chance of having these specific dll's somewhere on the system is very slim.
So it can find the file, but it cannot find the dependencies.

[image]

The previous version of the **tools-oss-suite** ([link](https://github.com/FPGAwars/tools-oss-cad-suite/releases/tag/v0.0.6)) did not include _iverliog_ or the _ivl_ folder. In that case you need to have **toolchain-iverilog**. In that package the system.vpi does not depend on the mentioned dll's !  

My solution (or hack) is to replace the content of the **tools-oss-cad-suite**-folder with the version 0.0.6. (download the tar.gz from github, rename the old folder, and extract the archive in a new folder with the original name). I did not find an apio command that let's you 'install' an older toolchain, or its not documented.

Maybe someone can make a rebuild of the tools-oss-cad-suite for windows and include the required dll-files ? 
Note: The **vpp.exe** file also needs some of them, and a library called _libreadline8.dll_. 

Best,

0xECEAD, 2022-04-17:
The missing dll can be downloaded [here](https://www.dll-files.com/libbz2-1.dll.html).
Place the x64 version in the **tools-oss-cad-suite\lib\** folder.

Obijuan, 2022-04-24:
Thanks for the feedback. The library libbz2 has already been included in the latest tools-oss-cad-suite apio package (version 0.0.8). Please, feel free to reopen this issue if the problem remains

closed 2022-04-24 by Obijuan