FPGAwars/apio #288: apio time y apio sim no funcionan

edsoncm22, 2022-02-24:
Error al ejecutar apio sim Windows 11

C:\Users\ABC: No such file or directory
error: Failed to open 'C:\Users\ABC~1\APIO~1\packages\TOOLS-~2\lib\ivl\system.vpi' because:
: No se puede encontrar el módulo especificado.

No top level modules, and no -s option.
scons: *** [PID_buck_tb.out] Error 1

edsoncm22, 2022-02-24:
Error al ejecutar apio time

Error: package 'ice40' is not installed
Please run:
   apio install ice40

mi pregunta es, no se supone que con apio 0.8 ya no es necesario el paquete ice40?

estos comando fueron ejecutados con apio 0.8 en Windows 11

Obijuan, 2022-05-01:
Este bug ya está solucionado en la version 0.8.1. La transición a las nuevas toolchains (oss-cad-suite) se ha completado con esta versión

Comprueba que apio sim y apio time te funciona bien. Si detectas errores adicionales, abre una nueva issue
Muchas gracias por el feedback

closed 2022-05-01 by Obijuan

edsoncm22, 2022-05-02:
Listo, probado en W11, funciona a la perfección, incluso con OSS-CAD-SUITE aumenta considerablemente la frecuencia máxima de funcionamiento de los diseños. 

Yo trabajo en el área de sistemas de control y tengo algunos proyectos de controladores utilizando la placa upduino v2 y empleando los bloques DSP. Pronto compartiré algún proyecto.

Obijuan, 2022-05-02:
Estupendo! Muchas gracias @edsoncm22 por las pruebas! 🙂️