FPGAwars/apio #291: request support for Zybo Zynq 7000

dsw, 2022-05-23:
I have a Zybo Zynq 7000.  If that is not easily supported by your tools, can you please provide a recommendation for any open source tools which might support it?

sergeyu, 2023-07-29:
Would be grateful to get such support for Zybo Z7-20.

zapta, 2024-10-15:
This board uses a Xiling FPGA.

cavearr, 2026-09-16:
@dsw @sergeyu  

This one can be closed as done: the Zybo Z7 is supported today. Apio 1.6.x carries the Xilinx architecture through the openxc7 package, and the definitions have both boards — `zybo-z7-10` (xc7z010) and `zybo-z7-20` (xc7z020)

Using the Zynq PL-only flow, where the bitstream configures the programmable logic over JTAG (no PS / Linux side involved). 

Anyone landing here: `apio boards | grep zybo`, then the usual `apio build` / `apio upload`.

cc: @zapta

closed 2026-09-16 by cavearr