FPGAwars/apio #294: your APIO install instructions just do not work

dsw, 2022-05-24:
Something essential is missing from your install instructions.

From your apio/README.md :

> # Installation
> 1. Install [Python](https://www.python.org/downloads) and [pip](https://pip.pypa.io)
> 2. Install the latest apio: ```pip install -U apio```

Running pip returns success.  So is apio installed?

As far as I can tell from reading the python pip documentation, pip does not seem to install any files in my $PATH, only in the python packaging system.  Yet your documentation speaks of a program "apio" that is supposed to exist in my path after running pip.

From your use guide, it seems that there should be a program called "apio" somewhere, as this documentation tells the user to just run it, as if it should be available in their $PATH: https://apiodoc.readthedocs.io/en/stable/source/user_guide/index.html

> Usage
> apio [OPTIONS] COMMAND [ARGS]
> You can execute just apio to see the help:
> $ apio
> Usage: apio [OPTIONS] COMMAND [ARGS]...

Yet, there is no file "apio" in my local tree or in my $PATH:
```
~/apio$ find . -iname apio -type f
~/apio$

$ apio
No command 'apio' found,
```

I am using the latest versions of python:
```
$ python --version
Python 2.7.18
$ python3 --version
Python 3.10.4

```

Could you please review your install instructions.  Would very much like to use apio.

Obijuan, 2023-02-05:
You will find the latest instructions on the wiki:

https://github.com/FPGAwars/apio/wiki

We still have to remove outdated documentation and point to the right one

Obijuan, 2024-04-03:
Documentation updated

closed 2024-04-03 by Obijuan