FPGAwars/apio #296: apio verify/build fails due to missing scons

fishnux, 2022-06-05:
Hi,

I installed the latest version of apio (through pip I believe) on MacOS, and also did `apio install --all`, which installed oss-cad-suite

It seems scons is obsolete/deprecated, but doing `apio verify` it still looks for scons. I've attempted to install scons anyway and had another error, so I wonder, do we still need scons?

Curiously, Icestudio is able to build without any problems - does it use `apio build` behind-the-hood?

jdclav, 2023-01-22:
I have come across something similar. I am not sure what error you were getting but when I verify I don't have any issues, however when I try to build, I get an error where scons is losing for a .json file for target hardware.config and I am not sure what the issue is.

Obijuan, 2023-02-05:
It would be great if you could test it again with the latest apio development version (0.8.2). You can install it with this command:

* **Linux** and **Mac**:

```
sudo pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
```
* **Windows**:
```
pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
```

jdclav, 2023-02-06:
My particular error was because I did not have a .lpf file. I did not realize that the ecp5 needed a .lpf file rather than a .pcf file. I am not sure if the original post was having the same issue.

Obijuan, 2023-02-06:
Thanks @jdclav 
@fishnux : I will wait for you tests with the latest apio version (0.8.2) to see if the problem is still there

fishnux, 2023-02-09:
@Obijuan seems to work fine now, though it's been a long time so I don't have the original scenario to reproduce, and I already have scons on my PATH from my previous attempt I guess. I think we can close this issue and re-open if someone experiences the same problem

closed 2023-02-09 by Obijuan

eroomde, 2023-09-07:
For what it's worth, I had the same issue as above with 0.8.2, I then tried:

```> apio install scons
(DEBUG) Profile path: /Users/ed/.apio/profile.json
(DEBUG) Home_dir: /Users/ed/.apio
Error: no such package 'scons'```

No luck, so I tried `brew install scons` and it works. I leave this here incase it helps any other lonely google sailors.

Parth-V-J, 2025-07-22:
"hi im new to this whole github stuf so bare with me and any help is welcome:"
here is the problem 
i installed apio, then tried to build some led program and this error occurred.
i tried to update scons and re downloaded the apio but problem persists.
help

```
yosys -p "synth_ice40 -top main -json hardware.json" -q and_gate.v
ERROR: Module `main' not found!
scons: *** [hardware.json] Error 1
```

zapta, 2025-07-22:
@Parth-V-J, most likely you are missing 'top-module' in apio.ini so apio assumes that the name of your top level verilog module in 'main'. You can see how it works in this example https://github.com/FPGAwars/apio-examples/blob/master/examples/alhambra-ii/ledon/apio.ini

If this doesn't solve your program, please provide here your apio version (run `apio --version`) and a copy of your `apio.ini` file.

If your version is greater than 0.9.5, you can find relevant docs here https://fpgawars.github.io/apio/project-file/#top-module-required

zapta, 2025-07-22:
@Parth-V-J, for future problems, please feel free to use the New Issue button or to ask in the Discussion section here.