FPGAwars/apio #297: leds.pcf file maps LEDs to nonexistent pins on Upduino3

jlittle22, 2022-07-03:
Current `leds.pcf`:

`set_io LED4 99
set_io LED3 98
set_io LED2 97
set_io LED1 96
set_io LED0 95
`
`[Sun Jul  3 11:35:12 2022] Processing upduino3
--------------------------------------------------------------------------------
nextpnr-ice40 --up5k --package sg48 --json hardware.json --asc hardware.asc --pcf leds.pcf -q
ERROR: package does not have a pin named '99' (on line 1)
ERROR: Loading PCF failed.
0 warnings, 2 errors
scons: *** [hardware.asc] Error 255
`

zapta, 2024-10-15:
@jlittle22, where did you find that leds.pcf file with reference to pin 99?

jlittle22, 2024-10-15:
Holy cow this is from a long time ago.

I don't remember the details here, but I just ran the basic setup commands and the `leds.pcf` file came with the repo that I cloned.

zapta, 2024-10-15:
Thanks John,  I can't find it in the apio provided examples so will close as can't-reproduce.  https://github.com/FPGAwars/apio-examples.

closed 2024-10-15 by zapta