FPGAwars/apio #298: Support for Gowin boards

k0055t, 2022-08-09:
Gowin FPGA chips have open source toolchain support. Boards as sipeed tang nano 9k are cheap and powerfull, but proprietary chinese IDE is drawback. Would it be possible to add support for them? They use seperate branch of nextpnr. There is article about using them with open source toolchain: https://www.geeklan.co.uk/?p=2919

wireboy5, 2024-08-06:
Hi @k0055t,  I have an iCESugar-nano that I have been happily using with apio, and I just recently purchased a Sipeed Tang Nano 20k. so I decided to take a crack at this.

Luckily, `nextpnr-himbaechel` is now no longer on a separate branch, and supports some of the Gowin boards via Apicula. It is also included in oss-cad-suite, which is a huge plus. In theory, it should not be hard at all to add other Sipeed boards using Gowin FPGAs supported by Apicula, but I only have the Nano 20k to test with. The PR is #390 if you are interested.

Patronics, 2024-09-08:
@k0055t, based on the changes in wireboy5's PR, I've attempted to add support for the other boards in the Tang Nano lineup with PR #411. Unfortunately, I don't have any of the Tang Nano boards to test it with, but if you're able to test it with any of them I'd be happy to fix any issues you find!

zapta, 2024-09-08:
As a side note, the next release of apio will allow to have custom boards.json and fpgas.json in the project directory. If exist, they will superceed to stock ones. 

https://github.com/FPGAwars/apio/blob/154592df13bad0cc9409f6cd8ca4bca55810ae12/apio/resources.py#L79

Patronics, 2024-09-27:
Some additional follow-up now that some hardware to test with has arrived, I have confirmed that my pull request works as intended with the Tang Nano 9k :)

zapta, 2024-12-29:
The gowin architecture is now part of the apio standard automatic tests.  We use the example ``sipeed-tang-nano-4k/blinky``

If anybody wants to try the latest apio dev with gowin, instructions here https://github.com/FPGAwars/apio/wiki/Apio-development-environment#just-trying-the-latest-apio-dev-version , just use the gowin example instead.

zapta, 2025-05-08:
Apio (dev branch) now supports Gowin. Closing.

closed 2025-05-08 by zapta