FPGAwars/apio #299: Documentation incomplete for adding new boards

mkvenkit, 2022-09-21:
The documentation for adding a new board to apio seems incomplete:

https://github.com/FPGAwars/apio/wiki/Adding-new-boards-to-apio

Please add some details. 

Thanks!

cavearr, 2022-09-22:
Hi @mkvenkit ! what type of programmer do you use? ftdi? jtag? Have you any documentation about your board?

mahesh-electronut, 2022-09-22:
Hi @cavearr thanks for your response. The RP2040 on the board acts as a programmer. Here's a guide to getting started with board. The board is not released yet, so I am still working on documentation and firmware:

https://github.com/mkvenkit/humble_ice/blob/main/getting_started.md

This is the Python script used to upload the bitstream:

https://github.com/mkvenkit/humble_ice/blob/main/hiprog.py

cavearr, 2022-09-22:
Thanks! mahesh, i think i could integrate without the needed to add to apio, because recently i integrate the badge for the mch2022 that uses the rp2040 as programmer like you. Give me some time to try and i'm telling you to check it.

One thing i need of you is the pcf file of your board.

mahesh-electronut, 2022-09-22:
@cavearr the only constraints required are the clk signal and the LED. So a minimal PCF file is:

```
set_io clk 35
set_io LED 13
```

The rest can be set by the user as needed.

cavearr, 2022-09-22:
Hi Mahesh! in Icestudio is needed to assign the input and output pins for the board, in the near future the user could modify easily from the interface but for now, the board should do the initial preset to provide at Icestudio the information to configure inputs and outputs 

If other pins in your board could be input/output, setup in this way in the pcf , but this, is needed. 

Enviados: Jueves, 22 de Septiembre 2022 8:21:23

mahesh-electronut, 2022-09-22:
OK, I will work on this and get back to you. Thanks

Obijuan, 2024-04-03:
The documentation on how to add boards to apio has been added. It is located here: https://github.com/FPGAwars/apio/wiki/Adding-new-boards-to-apio

closed 2024-04-03 by Obijuan