FPGAwars/apio #302: Got an unrecognized status code '404' when downloaded

aitesam961, 2022-09-29:
The "apio install --all" returns 404 error code when trying to install toolchain packages.
I am using latest release "apio-0.8.1" .
Please check the log below

C:\Users\aites>apio install --all
(DEBUG) Profile path: C:\Users\aites\.apio\profile.json
(DEBUG) Home_dir: C:\Users\aites\.apio
←[36mInstalling drivers package:←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
Download tools-drivers-windows-1.1.0.tar.gz←[0m
←[31mError: No module named 'termios'←[0m
(DEBUG) Profile path: C:\Users\aites\.apio\profile.json
(DEBUG) Home_dir: C:\Users\aites\.apio
←[36mInstalling examples package:←[0m
Download apio-examples-0.0.34.zip←[0m
←[31mError: package not availabe for this platform←[0m
(DEBUG) Profile path: C:\Users\aites\.apio\profile.json
(DEBUG) Home_dir: C:\Users\aites\.apio
←[36mInstalling gtkwave package:←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
Download tool-gtkwave-windows-3.3.77.tar.gz←[0m
←[31mError: No module named 'termios'←[0m
(DEBUG) Profile path: C:\Users\aites\.apio\profile.json
(DEBUG) Home_dir: C:\Users\aites\.apio
←[36mInstalling oss-cad-suite package:←[0m
Download tools-oss-cad-suite-windows_amd64-0.0.8.tar.gz←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
←[31mError: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/tools-oss-cad-suite/releases/download/v0.0.8/tools-oss-cad-suite-windows-0.0.8.tar.gz←[0m
[image]

aitesam961, 2022-09-29:
Was Following @ShawnHymel FPGA series to install. Tried installing on older release 0.6.7 as well but no luck. Though a little different error log as follows:

C:\Users\aites>apio install --all
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36mdfu←[0m package:
Download toolchain-dfu-windows_amd64-2020.11.24.tar.gz←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
←[31mError: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/toolchain-dfu/releases/download/v2020.11.24/toolchain-dfu-windows-2020.11.24.tar.gz←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36mdrivers←[0m package:
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
Download tools-drivers-windows-1.1.0.tar.gz←[0m
←[31mError: No module named 'termios'←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36mecp5←[0m package:
Download toolchain-ecp5-windows_amd64-2019.12.12.tar.gz←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
←[31mError: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/toolchain-ecp5/releases/download/v2019.12.12/toolchain-ecp5-windows-2019.12.12.tar.gz←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36mexamples←[0m package:
Download apio-examples-0.0.34.zip←[0m
←[31mError: package not availabe for this platform←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36mfujprog←[0m package:
Download toolchain-fujprog-windows_amd64-2020.10.6.tar.gz←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
←[31mError: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/toolchain-fujprog/releases/download/v2020.10.6/toolchain-fujprog-windows-2020.10.6.tar.gz←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36mgtkwave←[0m package:
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
Download tool-gtkwave-windows-3.3.77.tar.gz←[0m
←[31mError: No module named 'termios'←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36mice40←[0m package:
Download toolchain-ice40-windows_amd64-2019.12.11.tar.gz←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
←[31mError: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/toolchain-ice40/releases/download/v2019.12.11/toolchain-ice40-windows-2019.12.11.tar.gz←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36micesprog←[0m package:
Download toolchain-icesprog-windows_amd64-1.0.0.tar.gz←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
←[31mError: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/toolchain-icesprog/releases/download/v1.0.0/toolchain-icesprog-windows-1.0.0.tar.gz←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36miverilog←[0m package:
Download toolchain-iverilog-windows_amd64-1.1.1.tar.gz←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
←[31mError: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/toolchain-iverilog/releases/download/v1.1.1/toolchain-iverilog-windows-1.1.1.tar.gz←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36mscons←[0m package:
Download scons-3.0.1.tar.gz←[0m
←[31mError: package not availabe for this platform←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36msystem←[0m package:
Download tools-system-windows_amd64-1.1.2.tar.gz←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
←[31mError: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/tools-system/releases/download/v1.1.2/tools-system-windows-1.1.2.tar.gz←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36mverilator←[0m package:
Download toolchain-verilator-windows_amd64-1.0.0.tar.gz←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
←[31mError: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/toolchain-verilator/releases/download/v1.0.0/toolchain-verilator-windows-1.0.0.tar.gz←[0m
C:/Users/aites/.apio
C:/Users/aites/.apio
Installing ←[36myosys←[0m package:
Download toolchain-yosys-windows_amd64-2019.12.11.tar.gz←[0m
←[33mWarning: full platform does not match: windows_amd64                ←[0m
←[33m         Trying OS name: windows                ←[0m
←[31mError: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/toolchain-yosys/releases/download/v2019.12.11/toolchain-yosys-windows-2019.12.11.tar.gz←[0m

aitesam961, 2022-09-29:
Running on Core-i7 8650U, 8GB
Windows 10 Pro (21H2)
Build 19044.2075

Python 3.10

aitesam961, 2022-09-30:
Installed iCE Studio to see if the toolchain works that way. Will post shortly

aitesam961, 2022-09-30:
[image]
I'm stuck here for last 30 mins or so

Obijuan, 2023-02-05:
Now sure what is the problem. It would be great if you could test it again with the latest development version. You can install it with this command (in Windows):

```
pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
```

Obijuan, 2023-09-30:
A new apio (0.8.2) has been released. Could you please try it again? Thanks

Obijuan, 2024-03-26:
Please test it again with the new stable apio 0.9.2. If the error remains, please reopen this issue

closed 2024-03-26 by Obijuan