FPGAwars/apio #305: Apio does not recognize my icestick, claims "board icestick not connected"

dburr, 2022-11-06:
apio 0.8.1
Ubuntu 20.04.5 LTS

Subject line says it all. I am able to successfully build, verify, and sim (using the icestick/leds example project) however when I attempt to program the board, I get

```
(.apio_env) [2282] dburr@SDF-1 leds(master) $ apio upload
(DEBUG) Profile path: /home/dburr/.apio/profile.json
(DEBUG) Home_dir: /home/dburr/.apio
(DEBUG) Profile path: /home/dburr/.apio/profile.json
(DEBUG) Home_dir: /home/dburr/.apio
(DEBUG) Run Command: lsusb
(DEBUG) System_base_dir: /home/dburr/.apio/packages/tools-oss-cad-suite
(DEBUG) System bin dir: /home/dburr/.apio/packages/tools-oss-cad-suite/bin
(DEBUG) Executable file: /home/dburr/.apio/packages/tools-oss-cad-suite/bin/lsusb
(DEBUG) Profile path: /home/dburr/.apio/profile.json
(DEBUG) Home_dir: /home/dburr/.apio
(DEBUG) Run Command: lsftdi
(DEBUG) System_base_dir: /home/dburr/.apio/packages/tools-oss-cad-suite
(DEBUG) System bin dir: /home/dburr/.apio/packages/tools-oss-cad-suite/bin
(DEBUG) Executable file: /home/dburr/.apio/packages/tools-oss-cad-suite/bin/lsftdi
Error: board icestick not connected
```

`apio system --lsftdi` shows the board correctly:

```
(.apio_env) [2285] dburr@SDF-1 leds(master) $ apio system --lsftdi
(DEBUG) Profile path: /home/dburr/.apio/profile.json
(DEBUG) Home_dir: /home/dburr/.apio
(DEBUG) Run Command: lsftdi
(DEBUG) System_base_dir: /home/dburr/.apio/packages/tools-oss-cad-suite
(DEBUG) System bin dir: /home/dburr/.apio/packages/tools-oss-cad-suite/bin
(DEBUG) Executable file: /home/dburr/.apio/packages/tools-oss-cad-suite/bin/lsftdi
Number of FTDI devices found: 1
Checking device: 0
Manufacturer: FTDI, Description: Dual RS232-HS
```

And it also shows up in `lsusb`:

```
(.apio_env) [2286] dburr@SDF-1 leds(master) $ lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[...]
Bus 003 Device 003: ID 0bda:b720 Realtek Semiconductor Corp. 802.11n WLAN Adapter
Bus 003 Device 079: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
Bus 003 Device 002: ID 0a5f:0088 Zebra ZTC GX430t
[...]
```

Udev rules look fine, and I am a member of both `plugdev` and `dialout` groups.

I've tried completely nuking apio (including my user config directory) and reinstalling; no change. I've tried setting up apio in its own python virtualenv (in case some other python package is clashing with it); same behavior. I have also tried using this board on a different computer (a MacBook running OS X Big Sur 11.7.1, also running the latest version of apio (0.8.1) with the exact same behavior.

This is probably a fairly new revision of the board -- they have been out of stock at Digikey for a long time, but they just recently got some in stock, so I assume this is new stock. Maybe they changed something in the newer versions of this board?

dburr, 2022-11-09:
Upon further investigation, it appears that Lattice changed the `ftdi.desc` of this board. In `boards.json` if I change `icestick.ftdi.desc` to `Dual RS232-HS` then I am able to successfully upload.

steka, 2022-12-04:
> Upon further investigation, it appears that Lattice changed the `ftdi.desc` of this board. In `boards.json` if I change `icestick.ftdi.desc` to `Dual RS232-HS` then I am able to successfully upload.

Thanks for that investigation, it helped me.

closed 2022-12-26 by Obijuan

avielbr, 2023-05-12:
> Upon further investigation, it appears that Lattice changed the `ftdi.desc` of this board. In `boards.json` if I change `icestick.ftdi.desc` to `Dual RS232-HS` then I am able to successfully upload.

Hi, where is the `boards.json` located for you? Can't seem to find it...

avielbr, 2023-05-12:
> Upon further investigation, it appears that Lattice changed the `ftdi.desc` of this board. In `boards.json` if I change `icestick.ftdi.desc` to `Dual RS232-HS` then I am able to successfully upload.

Thanks for this fix! For anyone else having trouble finding the `boards.json` file, I found mine under `/home/$USER/.local/lib/python3.10/site-packages/apio/resources`

tempest1140, 2023-07-30:
> Upon further investigation, it appears that Lattice changed the `ftdi.desc` of this board. In `boards.json` if I change `icestick.ftdi.desc` to `Dual RS232-HS` then I am able to successfully upload.

Thank you! In case anyone still can't find their path after @avielbr 's comment, I found mine under:
C:\Users\$USER\AppData\Local\Programs\Python\Python311\Lib\site-packages\apio\resources

PR #308 (2022-12-26): Update ftdi id regex for icestick, as recent board has Dual RS232 HS…
Este cambio resuelve #305 .

Parece ser que las icestick recientes (dic-2022) se registran con la descripción "Dual RS232-HS". 

Translation: 
Resolves #305. It seems that recent icestick boards use "Dual RS232-HS" as description text.