FPGAwars/apio #306: oss-cad-suite can not be downloaded when looking for 'linux'

waterwin, 2022-12-16:
apio install oss-cad-suite
/usr/local/lib/python3.9/dist-packages/requests/__init__.py:102: RequestsDependencyWarning: urllib3 (1.26.5) or chardet (5.1.0)/charset_normalizer (2.0.12) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({})/charset_normalizer ({}) doesn't match a supported "
(DEBUG) Profile path: /home/erwin/.apio/profile.json
(DEBUG) Home_dir: /home/erwin/.apio
Installing oss-cad-suite package:
Warning: full platform does not match: linux_i686                
         Trying OS name: linux                
Error: Got an unrecognized status code '404' when downloaded https://github.com/FPGAwars/tools-oss-cad-suite/releases/download/v0.0.8/tools-oss-cad-suite-linux-0.0.8.tar.gz

Obijuan, 2023-02-05:
It seem that there are some depencencies missing in your system. Could you please try with the latest apio development version? (apio 0.8.2). You can install it with this command:

* **Linux** and **Mac**:

```
sudo pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
```
* **Windows**:
```
pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
```

waterwin, 2023-02-06:
As far as I can tell no change from previous situation.
***@***.***:~/Documents/apio$ lsapio.ini  icestick  leds.pcf  leds_tb.gtkw  leds_tb.v  ***@***.***:~/Documents/apio$ apio verify(DEBUG) Profile path: /home/erwin/.apio/profile.json(DEBUG) Home_dir: /home/erwin/.apioError: package 'oss-cad-suite' is not installedPlease run:   apio install oss-cad-suite

Installed on a clean Ubuntu VM (Oracle VirtualBox) that has as extra only pip and git installed

Erwin

 
 

   
sudo pip3 install -U ***@***.***#egg=apio
   
   - Windows:
pip3 install -U ***@***.***#egg=apio

Obijuan, 2023-09-30:
A new apio has been released (0.8.2). Could you please try it again?

Obijuan, 2024-03-26:
As there were no answer from the previous question, I think that is has been solved
Please, reopen it if the problem remains for you

closed 2024-03-26 by Obijuan