FPGAwars/apio #311: Alternative FTDI Configuration for iCE40-HX8K Breakout Board

lachjames, 2023-01-14:
Hi :) I just bought an iCE40-HX8K (my first FPGA, I'm still learning!) and I had trouble getting it to work with IceStudio. It seems that the board I purchased came with a simple USB cable, rather than the fancy FTUSB cable - not entirely sure what's going on, but eventuallyI traced the issue to the configuration file "boards.json", where

```
  "iCE40-HX8K": {
    "name": "iCE40-HX8K Breakout Board",
	...
    "ftdi": {
      "desc": "Lattice FTUSB Interface Cable"
    }
  },
```

needs to become

```
  "iCE40-HX8K": {
    "name": "iCE40-HX8K Breakout Board",
	...
    "ftdi": {
      "desc": "Dual RS232-HS"
    }
  },
```

for my board to work.

I'm not familiar enough with this pipeline to propose changing anything myself, and certainly not enough to submit a patch, but I figured this might help someone else?

bibble235, 2023-01-15:
Many many thanks for this as it helped me.  Looked in wrong directory at first but got there. Lost 3 hours of free time which is as everyone knows the most precious commodity. Could have been a lot more.

For anyone else I have a  ICE40UP5K-B-EVN from digi-key. This came through as 
Manufacturer: FTDI, Description: USB <-> Serial Converter

To fix this I had to change ~/.local/lib/python3.10/site-packages/apio/resources/boards.json to have

```
  "Bill": {
    "name": "Bills Board",
    "fpga": "iCE40-UP5K-SG48",
    "programmer": {
      "type": "iceprog"
    },
    "usb": {
      "vid": "0403",
      "pid": "6010"
    },
    "ftdi": {
      "desc": "USB <-> Serial Converter"
    }
  },
```
And upload went fine.

Obijuan, 2023-02-05:
It was fixed in the current apio development version (0.8.2). It would be great if you could install it and test that your board are working ok

You can install it with this command:

* **Linux** and **Mac**:

```
sudo pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
```
* **Windows**:
```
pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
```

bibble235, 2023-02-06:
Tried the suggested fix. I overwrite my old json but the result did work. Thanks.

Obijuan, 2023-09-30:
Apio 0.8.2 has been released. This bug should be fixed now. Feel free to open it again if it does not work

closed 2023-09-30 by Obijuan