FPGAwars/apio #312: iCESugar-Pro board definition appears incorrect

ahaberlach, 2023-01-30:
The iCESugar-Pro board definition appears to specify the incorrect part.

apio currently has it specified as ECP5-LFE5U-25F-CABGA381

The schematic at https://github.com/wuxx/icesugar-pro/tree/master/schematic specifies it as LFE5U-25F-6B-G256C

I was able to get it working with apio by changing my boards.join to ECP5-LFE5U-25F-CABGA256

Obijuan, 2023-01-31:
Thanks! It would be great if you could pull-request it

ahaberlach, 2023-01-31:
Ok, PR created -- I'm relatively new to the github workflow, and I don't seem to be able to assign the PR (or this issue), but I suspect you can merge or approve a merge.

Obijuan, 2023-02-05:
Thanks for the PR. It would be great if you could test the development version (apio 0.8.2). For installing it use this command:

* **Linux** and **Mac**:

```
sudo pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
```
* **Windows**:
```
pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
```

Xelef2000, 2023-05-07:
I tested the pr and building works without problems.
But I had problems with the upload.
The newer version of the iCESugar-Pro use geehy apm32 for the jtag interface.

Obijuan, 2023-09-30:
@ahaberlach apio 0.8.2 has been released with your fix. Could you please check that your initial issue has been fixed?

closed 2023-09-30 by Obijuan