FPGAwars/apio #313: Where to put .gtkwaverc on windows?

Kedarius, 2023-01-31:
I am not able to fiqure out where to put .gtkwaverc for gtkwave to load on start. The official docs says that it can be in home directory, the apio sim command says:
(DEBUG) Home_dir: C:\Users\<username>\.apio
however if I put it there it does not work. I also tried to put it in the actual working directory of the project.

Also how can I update the gtkwave version to something newer? The version apio installs is like 3 years old. I did replace it in 
C:\Users\<username>\.apio\packages\tool-gtkwave 
and it works but I suppose that it is not the correct way how to do it....

Obijuan, 2024-03-26:
In the latest stable apio 0.9.2 the gtkwave has been updated. Could you please check it?
I have not use gtkwave on Windows, but I guess that you should place the .gtkwaverc in C:\users\<username> (apio just execute gtkwave. Place it in the same location where gtkwave (without apio) read it by default)

Obijuan, 2024-04-03:
I am closing this issue. Feel free to reopen it if the problem remains

closed 2024-04-03 by Obijuan

Kedarius, 2024-04-21:
Sorry, I've stopped using apio. While it is a great tool to get into FPGA development, I did progress to using the tools directly and also from CI/CD environment....