FPGAwars/apio #315: Apio fails to recognise official Lattice boards

BFarkascode, 2023-02-01:
Hi,
I have multiple recently acquired Lattice devboards that I am trying to interface with using apio on a Raspberry 400 (with Rasp OS 64).
Apio 0.8.1 , drivers and tools_oss_cad_suite install fine and all seems fine (build, sim, verify, etc.) until I attempt an upload. Then, apio fails to find neither my Icestick nor the Lattice HX8K_EVB.
I checked "apio system --lsftdi" and the FTDI chips are seen using both boards, albeit the manufacturer comes out as "FTDI" and the description as "FT232 Dual serial". Checking some successful apio runs on Youtube, it seems that lsftdi readout should actually provide manufacturer "Lattice" and description "Icestick" or "HX8K_EVB" instead to allow interaction.
Thanks in advance for looking into this!

BFarkascode, 2023-02-02:
Okay, so it was the same issue indicated here:
https://github.com/FPGAwars/apio/issues/311
The "boards.json" file needs to be updated manually for the Icestick and the HX8K, so the description apio will be looking for on the FTDI channel for these boards will be "Dual RS232-HS" instead of the original description.
Of note, it is a bit of a pain to find, where apio is installed, at least on the Rasp OS 64, using standard out-of-the-box python setup. For me, it was in "/usr/local/lib/python3.9/dist-packages/apio" and not in "site-packages" as indicated on the link above. At any rate, the path can also be found using  "python pip show apio".
The "boards.json" file will be in the "resources" folder within the "apio" folder.
Editing the file can only be done using sudo.

Obijuan, 2023-02-05:
It was fixed in the current Apio development version (0.8.2). It would be great if you could test it. For installing the development version use this command:
```
sudo pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
```

Check that the correct version was installed

```
$ apio --version
apio, version 0.8.2
```

Try `apio upload` to check if it is working

BFarkascode, 2023-02-10:
Hi Obijuan,

Thanks for the quick reaction!

I've just tested the dev version and I did not encounter the issue using an Icestick.

Cheers!

Obijuan, 2023-02-10:
ok, thanks for the test. I will close this issue. Feel free to open it again (or a new one) if you find problems with a specific board

closed 2023-02-10 by Obijuan