FPGAwars/apio #325: multiple files in one apio project

bendelathouwer, 2023-05-01:
Quick quistion, is it possible (in a multiple verilog source prodject) to verify and simulate one specific source file?
So for example when you have a clock devider and a shift register, is it posible to verify and simulate for example the clock divider alone?

Thanks for the help and with kind regards,
Ben

obtitus, 2024-09-15:
yes, use the `apio sim --testbench` argument, i.e.
```
apio sim --testbench clock_divider_tb.v
```

zapta, 2024-09-15:
Apropos apio testbenches, here is a general purpose macros file to facilitates apio friendly test benches.  It uses the apio injected macro INTERACTIVE_SIM to determine if to abort when assertion fails (when running test command) or continue (when running sim command). 

apio_testing.vh
```
// Utility macros for testbenches that are compatible with apio.

`default_nettype none

`define DUMP_FILE_NAME(x) `"x.vcd`"

// Call this macro at the begining of the testbench macro. It defines
// a clk signal and a variable clk_num that indicates the clock num.
`define DEF_CLK \
    reg clk = 0; \
    integer clk_num = 0; \
    always begin \
        #10 clk = ~clk; \
        if (clk) clk_num += 1; \
    end

// Asserts that the value of 'signal' is 'value'. If not, it prints an error message and aborts
// the simulation. When running under apio sim, the macro INTERACTIVE_SIM is defined
// and the failed assertions does not exist to allow showing the sigmulation results
// in the graphical window.
`define EXPECT(signal, value) \
    if (signal !== (value)) begin \
        $display("*** ASSERTION FAILED in %m (clk_num=%0d): expected (signal == value), actual: 'h%h", (clk_num), (signal)); \
        `ifndef INTERACTIVE_SIM \
             $fatal; \
        `endif \
    end

// Transition from clock low to clock high. Typically this is not
// called directly and clock is managed using `CLK().
`define CLK_HIGH \
    begin \
        `EXPECT(clk, 0); \
        @ (posedge clk); \
        #2; \
        `EXPECT(clk, 1); \
    end

// Transition from clock high to clock low. Typically this is not
// called directly and clock is managed using `CLK().
`define CLK_LOW \
    begin \
        `EXPECT(clk, 1); \
        @ (negedge clk); \
        #2; \
        `EXPECT(clk, 0); \
    end

// Simulate one clock. Wait for low to high and then high to low transition.
`define CLK \
    begin \
        `CLK_HIGH \
        `CLK_LOW \
    end

// Simulate n clocks.
`define CLKS(n) \
    begin \
        repeat(n) begin \
            `CLK \
        end \
    end

// Place this macro immediatly after the 'initial begin' statement of the testbench.
// 'testbench' is the name of the testbench module. The macro sets the file
// that will contains the simulation results.
// The macro VCD_OUTPUT is defined automatically by the apio commands sim and test
// and contains base name of the expected output file.
`define TEST_BEGIN(testbench) \
    begin \
        `ifdef VCD_OUTPUT \
            $dumpfile(`DUMP_FILE_NAME(`VCD_OUTPUT)); \
            $dumpvars(0, testbench); \
        `else \
            $fatal(1, "Automatic macro VCD_OUTPUT not defined."); \
        `endif \
    end

// Place this macro at the end of the 'initial begin' block of the testbench.
`define TEST_END \
    begin \
        @ (posedge clk); \
        $display("End of simulation"); \
        $finish; \
    end
```

Sample testbench file:
```
// An example of an automated test. Run 'apio test' to test and
// 'apio sim' to examine the wave signals.

`include "apio_testing.vh"

`timescale 10 ns / 1 ns

// A macro to increment the tested counter by n.
`define INCREMENT(n) \
    begin \
        `CLK \
        count_en = 1; \
        `CLKS(n) \
        count_en = 0; \
        `CLK \
    end

module main_tb ();
  // Declare the 'signal' name that is managed by apio_testings.vh.
  `DEF_CLK;

  // Inputs to the tested module.
  reg reset;
  reg count_en;

  // Outputs of the tested module
  wire [3:0] digit_1;
  wire [3:0] digit_10;
  wire carry;

  // A utility signal that indicates the number of clocks on which
  // the carry output of the tested module was high.
  reg [5:0] carry_count = 0;

  always @(posedge clk) begin
    carry_count <= reset ? 0 : carry ? carry_count + 1 : carry_count;
  end

  // The instance of the tested module.
  main tested (
      .clk(clk),
      .reset(reset),
      .count_en(count_en),
      .digit_1(digit_1),
      .digit_10(digit_10),
      .carry(carry)
  );

  // The test sequence.
  initial begin
    `TEST_BEGIN(main_tb)

    // Reset the tested module.
    count_en = 0;
    reset = 1;
    `CLKS(3)
    reset = 0;
    `EXPECT(digit_10, 0)  // count = 00
    `EXPECT(digit_1, 0)
    `EXPECT(carry_count, 0)
    `CLKS(3);

    // Count to 09.
    `INCREMENT(9)
    `EXPECT(digit_10, 0)
    `EXPECT(digit_1, 9)
    `EXPECT(carry_count, 0)

    // Test transition from 09 to 10,
    `INCREMENT(1)
    `EXPECT(digit_10, 1)  // count = 10
    `EXPECT(digit_1, 0)
    `EXPECT(carry_count, 0)

    // Count to 99
    `INCREMENT(89)
    `EXPECT(digit_10, 9)  // count = 99
    `EXPECT(digit_1, 9)
    `EXPECT(carry_count, 0)

    // Test transition from 99 to 00.
    `INCREMENT(1)
    `EXPECT(digit_10, 0)  // count = 00
    `EXPECT(digit_1, 0)
    `EXPECT(carry_count, 1)

    // Count to 12.
    `INCREMENT(12)
    `EXPECT(digit_10, 1)  // count = 12
    `EXPECT(digit_1, 2)
    `EXPECT(carry_count, 1)

    // All done.
    `CLKS(10)
    `TEST_END
  end

endmodule

```

zapta, 2024-10-15:
Question answered. Closing.

closed 2024-10-15 by zapta