FPGAwars/apio #326: Error: board icestick not connected

neo-seeker, 2023-05-25:
Hi Folks. I got myself a new ICE40 Eval board iCE40HX1K-TQ144. Used APIO repository and when I run the command" apio upload, I get the following:

(APIO) PS C:\apio\icestick\leds> apio upload
c:\program files\python38\lib\site-packages\requests\__init__.py:102: RequestsDependencyWarning: urllib3 (1.26.15) or chardet (5.1.0)/charset_normalizer (2.0.12) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({})/charset_normalizer ({}) doesn't match a supported "
(DEBUG) Profile path: C:\Users\singh\.apio\profile.json
(DEBUG) Home_dir: C:\Users\singh\.apio
(DEBUG) Profile path: C:\Users\singh\.apio\profile.json
(DEBUG) Home_dir: C:\Users\singh\.apio
(DEBUG) Run Command: lsusb
(DEBUG) System_base_dir: C:\Users\singh\.apio\packages\tools-oss-cad-suite
(DEBUG) System bin dir: C:\Users\singh\.apio\packages\tools-oss-cad-suite\bin
(DEBUG) Executable file: C:\Users\singh\.apio\packages\tools-oss-cad-suite\bin\lsusb.exe
(DEBUG) Profile path: C:\Users\singh\.apio\profile.json
(DEBUG) Home_dir: C:\Users\singh\.apio
(DEBUG) Run Command: lsftdi
(DEBUG) System_base_dir: C:\Users\singh\.apio\packages\tools-oss-cad-suite
(DEBUG) System bin dir: C:\Users\singh\.apio\packages\tools-oss-cad-suite\bin
(DEBUG) Executable file: C:\Users\singh\.apio\packages\tools-oss-cad-suite\bin\lsftdi.exe
**Error: board icestick not connected**

I have installed the Zadig drivers too. Tried Ubuntu and and windows but still to no avail.

Obijuan, 2023-05-26:
Please, try with the apio development version
You can install it with this command:
pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio
(Make sure git is installed before)

neo-seeker, 2023-05-26:
Thanks Obijuan. It helped. But as you suggested that I had selected the wrong board. I think the examples for icestick would also not apply therefore. After running the "apio upload" command, I get the following error:  

(APIO) PS C:\apio\icestick\leds> apio upload
C:\Program Files\Python38\lib\site-packages\requests\__init__.py:102: RequestsDependencyWarning: urllib3 (1.26.15) or chardet (5.1.0)/charset_normalizer (2.0.12) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({})/charset_normalizer ({}) doesn't match a supported "
C:/Users/singh/.apio
[Fri May 26 15:18:38 2023] Processing iCE40-HX1K-EVB
------------------------------------------------------------------------------------------------------------------------
yosys -p "synth_ice40 -json hardware.json" -q leds.v
nextpnr-ice40 --hx1k --package vq100 --json hardware.json --asc hardware.asc --pcf leds.pcf -q
ERROR: package does not have a pin named '98' (on line 10)
ERROR: Loading PCF failed.
0 warnings, 2 errors
scons: *** [hardware.asc] Error 4294967295
============================================= [ ERROR ] Took 1.85 seconds =============================================

Does this mean I need to look for examples elsewhere in APIO examples?

neo-seeker, 2023-05-26:
Hi Obijuan, thanks for your help. The board error got resolved. I followed the following steps for iceprogduino

To install iceprogduino:

clone the repository
git clone https://github.com/OLIMEX/iCE40HX1K-EVB.git

cd into the direcory
cd iCE40HX1K-EVB/programmer/iceprogduino

compile and install
make
sudo make install

Run iceprogduino
iceprogduino

Now I get the following issue with iceprogduino.

lisa@lisa-VirtualBox:~/apio/iCE40-HX1K-EVB/leds$ iceprogduino -I /dev/bus/usb/001/005 hardware.bin 

**error 25 from tcgetattrerror 25 from tggetattrSerial: /dev/bus/usb/001/005: Inappropriate ioctl for device**

file size: 32220

erase 64kB sector at 0x000000..erased

programming..

prog 0x000000 +0x100........................................................................................................................................................................

The programming goes into endless loop with the aforementioned error.

wwoofbum, 2023-07-25:
I am getting the same error.  I am running apio on a Raspberry Pi 4B, using the latest 64 bit PiOS, followed the installation steps it the docs, followed the Quick Start guide, using an icestick with an FPGA labeled iCE40HX1K-TQ144

Obijuan, 2023-09-30:
Apio 0.8.2 has been released with a fix for this initial issue. Feel free to reopen it again if it does not work (or open a new issue for a different problem)

closed 2023-09-30 by Obijuan