FPGAwars/apio #329: weak pull up on icestick

MaxFlatline, 2023-08-10:
Hello
There seems to be an issue with building projects for icestick board. Unused LED pins seem to have a weak pull-up, and i haven't been able to switch it off. Probably a good idea to allow user to set desired unused output rules before building project.

MaxFlatline, 2023-08-29:
I have not found any solution. Probably will use bare oss-cad-suite with icestorm tools

Obijuan, 2023-09-30:
It is not a problem with the toolchains. Just set to 0 the unused leds in your verilog code
In higher level tools, like icestudio, a rule is provided to it automatically for the unused leds. If you are doind it directly in verilog you should do it manually

closed 2023-09-30 by Obijuan