FPGAwars/apio #335: Add board support for ShastaPlus (sku CAS-1111)

Thomas-J-Kidd, 2023-12-23:
We are looking to work with the ShastaPlus (sku CAS-1111) board called the WEB FPGA.

The webpage for the board can be found here: https://book.webfpga.io/
[image]

This is our schematic image that we were able to find
[image]

zapta, 2024-10-15:
Hi @Thomas-J-Kidd, we don't have a Shasta board to test. Can you provide the necessary board definitions?  You can see examples in file boards.json.  

https://github.com/FPGAwars/apio/blob/develop/apio/resources/boards.json

zapta, 2024-12-22:
Hi @Thomas-J-Kidd, the next apio release will support it via a custom boards.json in the project directory and once you validate it you will be able to contribute the board definition the apio project.

Closing. Please reopen if you have any questions.

closed 2024-12-22 by zapta

Thomas-J-Kidd, 2024-12-22:
Awesome thanks so much! Will look into this soon!