FPGAwars/apio #336: TinyFPGA-BX on Ubuntu 20.04: "Error: board TinyFPGA-BX not connected"

ll16mojh, 2023-12-31:
Hi,

I've followed the Ubuntu 20.04 Quick Start for the TinyFPGA-BX but when I run ```apio upload``` I get:

```
matt@matt-WE62-7RJ:~/TinyFPGA-BX/Blinky$ apio upload
(DEBUG) Profile path: /home/matt/.apio/profile.json
(DEBUG) Home_dir: /home/matt/.apio
(DEBUG) Profile path: /home/matt/.apio/profile.json
(DEBUG) Home_dir: /home/matt/.apio
(DEBUG) Run Command: lsusb
(DEBUG) System_base_dir: /home/matt/.apio/packages/tools-oss-cad-suite
(DEBUG) System bin dir: /home/matt/.apio/packages/tools-oss-cad-suite/bin
(DEBUG) Executable file: /home/matt/.apio/packages/tools-oss-cad-suite/bin/lsusb
Error: board TinyFPGA-BX not connected
```

After installing a fresh copy of Ubuntu 20.04 and running update, upgrade and autoremove, all I have done is:

```
sudo apt install python3
sudo apt install python3-pip
sudo pip3 install -U apio
apio install -a
apio drivers --serial-enable
```
At this point it said something about restarting, for the dialout group to work (I forget the exact wording) so I restarted my laptop. then ran:
```
sudo usermod -a -G dialout $USER
exec su -l $USER
sudo pip3 install tinyprog
```
At this point, as per quick start, I plugged in my board. One red LED is on continuously, the other is pulsing slowly which I believe means its in bootloader mode?
```
tinyprog --update-bootloader
apio examples -d TinyFPGA-BX/Blinky
cd TinyFPGA-BX/Blinky/
apio upload
```
And this gave the result above. At no point did i see any errors or anything that seemed to indicate a problem to me, so I'm a bit confused what I've done wrong or am missing (probably something simple and embarrassing!).

If I run ```tinyprog``` while its plugged in it returns:
```
    TinyProg CLI
    ------------
    Using device id 1d50:6130
    Only one board with active bootloader, using it.
```
and after removing the board ```tinyprog``` returns:
```
    TinyProg CLI
    ------------
    Using device id 1d50:6130
    No port was specified and no active bootloaders found.
    Activate bootloader by pressing the reset button.
```

Any suggestions welcome!

towalker, 2024-03-02:
Hello. I'm running into the same problem. The additional info is that I first tried using the  the suggested apio version in the guide (apio==0.4.0b5), but had problems building. Then, I installed the latest version (0.8.4), and could build but not upload. 
Running lsusb (not the one in /tools-oss-cad-suite/bin/, but the system's one) I can see "Bus 001 Device 011: ID 1d50:6130 OpenMoko, Inc. " 
Running "tynyprog -l" also gives this same device id, but says there is no metadata for the board. Anyone had the same problem?  I'm on Ubuntu 22.04.
 
>     TinyProg CLI
>     ------------
>     Using device id 1d50:6130
>     Only one board with active bootloader, using it.
>     Boards with active bootloaders:
> 
>         /dev/ttyACM0: No metadata
> 

And the result of uploading:

> (DEBUG) Profile path: /home/tweber/.apio/profile.json
> (DEBUG) Home_dir: /home/tweber/.apio
> (DEBUG) Profile path: /home/tweber/.apio/profile.json
> (DEBUG) Home_dir: /home/tweber/.apio
> (DEBUG) Run Command: lsusb
> (DEBUG) System_base_dir: /home/tweber/.apio/packages/tools-oss-cad-suite
> (DEBUG) System bin dir: /home/tweber/.apio/packages/tools-oss-cad-suite/bin
> (DEBUG) Executable file: /home/tweber/.apio/packages/tools-oss-cad-suite/bin/lsusb
> Error: board TinyFPGA-BX not connected

towalker, 2024-03-02:
Just additional info. In regard to my previous comment on not being able to "apio build" with apio==0.4.0b5, this could be fixed by changing the following: 

On file:  ~/.apio/packages/tool-scons/script/../engine/SCons/Util.py

Changed line 49, from:
`from collections import Iterable`
to:
`from collections.abc import Iterable`

Although this allowed me to use "apio build" using apio==0.4.0b5, running "apio upload" still does not recognize the board.

Obijuan, 2024-03-26:
Could you please try it again with the latest stable apio 0.9.2? (@ll16mojh and @towalker)
I've tested the tinyFPGABX in my ubuntu 22.04 and it work ok. I am not sure if it has been fixed for you

Obijuan, 2024-04-03:
I am closing this issue. Please feel free to reopen it if the problem remains

closed 2024-04-03 by Obijuan

wxyangf, 2024-05-07:
Was just trying to get started with an TinyFPGA-BX today and ran into the same issue -- will post logs tomorrow. Don't have access to that system ATM.

towalker, 2024-05-15:
Although I can "apio build" with apio 0.9.5 (after setting the "top-module" name in apio.ini) , I still can't get it to upload to the board.

```
$ apio upload
Activate bootloader by pressing the reset button
Error: board TinyFPGA-BX not connected

```
Pressing the reset button makes no change in the response. Probably it is a problem with the board itself, which is very unfortunate, as it costed me much more than the original price due to transportation and importation costs to Brazil.

towalker, 2024-05-15:
Hello, it seems the problem is related to missing metadata in boards shipped in 2023.

The unofficial workaround is discussed in [(https://github.com/tinyfpga/TinyFPGA-BX/issues/37)].