FPGAwars/apio #337: deactivating gtkwave

theogvn, 2024-01-27:
Hey, I'm not using GTKWave and thus have not install it in on my Mac, instead I'm using WaveTrace in VS code. But when I run `$ apio sim` I constantly get the error `sh: gtkwave: command not found` would it be possible to add a parameter to `sim` to deactivate the automatic start up of GTKWave ?

ghxstbox, 2024-01-27:
It's just checking to see if gtkwave is executable, which it shouldn't be, assuming that you don't have it on your Mac. If you wanted to deacticate the automatic startup of GTKWave, you may need to alter the functions of the script that elicit the sim command, or delete it entirely from the commands folder. It seems like there is some indication of the script compiling a list of all the python commands from the commands folder: https://github.com/FPGAwars/apio/blob/b0827f9ee277ca9f900f53e8110048be698cf252/apio/__main__.py#L61

theogvn, 2024-01-27:
yeah that's what I tried to do, but unfortunately I haven't been able to find the piece of code that tries to execute gtkwave from the shell. If someone knows ?

Obijuan, 2024-03-18:
from apio 0.0.9 there is a new command: `apio tst` that is similar to `apio sim`  but without launching gtkwave. Please try it (currently it is in the development branch, but it will be soon releases as a stable version)
[here you can fin the instructions for installing apio development](https://github.com/FPGAwars/apio/wiki/Apio-development-version)

zapta, 2024-03-18:
> from apio 0.0.9 there is a new command: apio test (thanks to @zapta

Thanks Juan. I have been using the dev branch for some time. Will pull the
latest version.

A few notes about the test command:

1. By default it runs all the testbenches (*_tb.v) while the sim runs just
one.

2. It's intended to be used with assertions that fail the test
automatically if something is unexpected. Similar to unit testing of
software.

3. When debugging a failed assertion, it's useful to run the sim command
with assertions disabled to examine what's going on. The sim command now
defines a macro called INTERACTIVE_SIM that allows the user to disable
assertions.

4. An example for a testbench with assertions is available here
https://github.com/FPGAwars/apio-examples/tree/master/upduino31/testbench  (you
don't need to have an Upduino board to try it)

Obijuan, 2024-03-26:
The stable apio 0.9.2 has been released, with the new apio test command. Please test it and if you find it useful, close this issue

Obijuan, 2024-04-03:
I am closing this issue. Please, feel free to reopen it if the problem remains

closed 2024-04-03 by Obijuan