FPGAwars/apio #340: Please support multiple test benches in the same directory.

zapta, 2024-02-19:
In my project i have mutliple verilog files foo.v, bar.v, and so on, and each has a testbench foor_tb.v, bar_tb.v, etc.  When I try to run 'apio sim' it complains that "more than one testbenches used" and pick one arbitrarily. 

Please extend 'apio sim' to accept the name of the testbench to run. Having multiple .v files with a test bench for each is a very common practice.

zapta, 2024-02-19:
Hi @Obijuan, can I send you a PR for this? It adds a --testbench flag to the sim command.  E.g. 

```
apio --teshbench my_module_tb.v
```

I have a working version with a patched 0.8.4.  The key patch is in SConstruct and looks like this:

```
# -- Get a list of all the verilog files in the src folfer, in ASCII, with
# -- the full path. All these files are used for the simulation
v_nodes = Glob('*.v')
v_files = [str(f) for f in v_nodes]

# Create lists of module and testbench files. Test benches are assumed
# to end with '_tb.v', case insensitive.
src_sim = [f for f in v_files if f[-5:].upper() != '_TB.V']
list_tb = [f for f in v_files if f[-5:].upper() == '_TB.V']

# Handle the testbench selection, if any. 
testbench = None

if TESTBENCH:
  testbench = TESTBENCH
else:
  if len(list_tb) > 1:
    print('Warning: more than one testbench found.')
  if len(list_tb) > 0:
    testbench = list_tb[0]
    
# Add the testbench to the list of compiled files.
if testbench:
  src_sim.append(testbench)
```

Obijuan, 2024-02-19:
Hi @zapta!

Yes, you can send the PR, for sure. Please, send it to the Develop branch

zapta, 2024-02-19:
Thanks @Obijuan. I sent you this PR https://github.com/FPGAwars/apio/pull/341

Obijuan, 2024-03-18:
Hi @zapta! I think that this issue is fixed with your PR. Please, close this issue if everything is working as expected. Thank you very much for all your contributions

zapta, 2024-03-18:
Closing. The sim command now has a flag to select the testbench to simulate.

```
$ apio sim --help
Usage: apio sim [OPTIONS]

  Launch the verilog simulation.

Options:
  -p, --project-dir path     Set the target directory for the project.
  -t, --testbench testbench  Specify the testbench file to simulate.
  -h, --help                 Show this message and exit.
```

closed 2024-03-18 by zapta

375432636, 2024-03-20:
it seems like not support windows version yet. 

```Powershell
PS Documents\project\fpga\step_baseboard_v40_v2> apio sim --help
Usage: apio sim [OPTIONS]

  Launch the verilog simulation.

Options:
  -p, --project-dir path  Set the target directory for the project.
  --help                  Show this message and exit.
PS Documents\project\fpga\step_baseboard_v40_v2> apio sim -t    
Usage: apio sim [OPTIONS]
Try 'apio sim --help' for help.

Error: No such option: -t
PS Documents\project\fpga\step_baseboard_v40_v2> apio --version
apio, version 0.8.4
PS Documents\project\fpga\step_baseboard_v40_v2>
```

zapta, 2024-03-20:
That was added post 0.8.4 so I presume it will come with the upcoming 0.9.

https://pypi.org/project/apio/#history