FPGAwars/apio #345: Please have apio installing gtkwave automatically.

zapta, 2024-02-21:
gtkwave is required for the 'apio sim' command. Please have apio installing it automatically, like the rest of the dependencies. This will simplify the installation of apio.

Obijuan, 2024-03-26:
gtkwave is included in the oss-cad-suite, so it should not be difficult to include it in apio. The gtkwave apio package could be deprecated in that case

zapta, 2024-10-26:
@Obijuan, @cavearr, a question for you. 

I am looking into automatic installation of gtkwave and graphviz/dot on all platforms and it seems that the upstream oss-cad-tools contains both. Any reason why not to include the entire upstream package in the apio tools-oss-cad-suitepackage?  

The compressed ratio is about 4:3 so not a big difference and the advantages are 

1) making the entire oss-cad-suite availabe to apio users (via the apio raw command) 
2) and avoiding the risk of missing out files and libraries (it's not obvious to figure out the dependencies of graphivz and gtkwave).
3) Simpler install script and upgrade path to future versions of oss-cad-suite.

If it's ok with you, I will go ahead and make the change.

```
/Users/user/projects/tools-oss-cad-suite-dev/repo $ ls -al _upstream/*/oss-cad-suite-*
-rw-r--r--  1 user  staff  414237486 Aug  1 19:21 _upstream/darwin/oss-cad-suite-darwin-x64-20240802.tgz
-rw-r--r--  1 user  staff  388950642 Aug  1 19:39 _upstream/darwin_arm64/oss-cad-suite-darwin-arm64-20240802.tgz
-rw-r--r--  1 user  staff  615296561 Aug  1 19:02 _upstream/linux_x86_64/oss-cad-suite-linux-x64-20240802.tgz
-rw-r--r--  1 user  staff  285861054 Aug  1 19:18 _upstream/windows_amd64/oss-cad-suite-windows-x64-20240802.exe

/Users/user/projects/tools-oss-cad-suite-dev/repo $ ls -al _packages/tools-oss-cad-suite-*
-rw-r--r--  1 user  staff  307364026 Oct 25 15:12 _packages/tools-oss-cad-suite-darwin-0.1.0.tar.gz
-rw-r--r--  1 user  staff  294146036 Oct 25 15:13 _packages/tools-oss-cad-suite-darwin_arm64-0.1.0.tar.gz
-rw-r--r--  1 user  staff  433528931 Oct 25 14:56 _packages/tools-oss-cad-suite-linux_x86_64-0.1.0.tar.gz
-rw-r--r--  1 user  staff  336623713 Oct 25 15:14 _packages/tools-oss-cad-suite-windows_amd64-0.1.0.tar.gz
```

cavearr, 2024-10-26:
Hello @zapta, this option is the preferred one for me. I raised it a long time ago but I didn't continue with this because I think @Obijuan prefers to have more controlled binaries. 

I updated the toolchain some time ago due to the needs of new features included in the new versions of yosys such as inout pins and although I did it in the wip versions, many collections stopped working, generating a big headache for @Obijuan. I think the way Icestudio is approaching it, at least at the moment, a "continuous release" mode like oss-cad-suite could be an issue, but I don't know if this stance has changed over time on obijuan's part .

At icestudio I am exploring being able to have different versions of celery simultaneously and not only being able to choose between stable and development. Perhaps in celery the option of choosing the mode (rolling-release, or specific versions of the toolchain) could be considered.

In other software such as npm there are package managers that could be similar to celery that allow just this, being able to say now I want to use version 22.04, tomorrow 13.01.....

For me, this is the preferred option (align with oss-cad-suite and have the complete suite or a version manager. For my part, I am exploring this from icestudio since I don't usually get too involved in celery except for specific things).

zapta, 2024-10-26:
@cavearr, my plan was to have something in the middle, still having the current apio package release process but with a more aggressive copying of files from upstream package to apio package. For example, instead of copying selected files form the ./bin directory, copy all the files from the bin directory.  This will still leave room to package tweaking when needed.

https://github.com/FPGAwars/tools-oss-cad-suite/blob/9ad567093164f7a194490503131fcff73a152882/scripts/install_darwin_arm64.sh#L7

cavearr, 2024-10-26:
@zapta, for mi go ahead! ;) very good job master!

zapta, 2024-10-26:
Thanks @cavearr.  I just run an experiment, replacing install_darwin_arm64.sh with the simplified script and so far it works well, including gtkwave and graphviz/dot.  Will test it more.  tools-oss-cad-suite package size went up from 262MB to 365MB.

https://github.com/FPGAwars/tools-oss-cad-suite/blob/main/scripts/install_darwin_arm64.sh

```
# Assert package_dir is empty
if [ "$(ls -A $PACKAGE_DIR)" ]; then
     echo "Error: [$PACKAGE_DIR] is not empty."
     exit 1
fi

# Copy key directories.
cp -rp $SOURCE_DIR/bin $PACKAGE_DIR
cp -rp $SOURCE_DIR/lib $PACKAGE_DIR
cp -rp $SOURCE_DIR/libexec $PACKAGE_DIR
cp -rp $SOURCE_DIR/share $PACKAGE_DIR

# -- Replace bin/nextpnr-ice40 with our simplified template.
install $TEMPLATE/nextpnr-ice40 $PACKAGE_DIR/bin
```

cavearr, 2024-10-26:
I'll check it tomorrow, thank you very much. The increase in size is acceptable with the functionality gained

zapta, 2024-11-12:
Implemented. Closing.

closed 2024-11-12 by zapta