FPGAwars/apio #346: [Feature Request] Add support for automatic testing.

zapta, 2024-02-21:
NOTE: If this feature request is reasonable, I can try implementing it and send PR requests.

The command ``apio sim``  is useful for examining and understanding logic behavior and signals but is not a good fit for automatic testing of multiple testbenches that contains assertions for correctness. First it runs just a single testbench and second the interactive ``gtkwave``  windows is a distraction.

This request is for a new command ``apio test`` with behavior along these lines:

1. It processes all the testbenches (independently, one after the other).
2. It does not activate the ``gtkwave`` interactive windows.
3. The command succeeds if all testbenches succeeded (e.g. no syntax error, and no call to $fatal).
4. The optional flag ``--testbench`` allows to run just a single testbench, for example, when developing a specific test.

Notes:
* Ok for the command to fail on the first testbench failure.
* Need to have a way to run a testbench with failing assertion with the ``apio sim`` command for debugging.  For example by injecting to the benchtests macros that indicates if the run is automatic or interactive.

Obijuan, 2024-02-21:
Yes, please, go ahead

zapta, 2024-02-22:
Ok, I will work on it. It may take time though.

Obijuan, 2024-02-22:
Thank you very much for all your contributions! 🙂️  I really appreciate them

Obijuan, 2024-04-03:
I am closing this issue as completed