FPGAwars/apio #348: Errors in apio build on examples

znmeb, 2024-02-22:
I'm using the directions at <https://pico-ice.tinyvision.ai/md_programming_the_fpga.html#autotoc_md35> and trying to build the iCE40-UP5K examples. The host is Windows Subsystem for Linux Ubuntu 22.04 LTS. See the attached script and log file.

What am I missing?

[apio-errors.zip](https://github.com/FPGAwars/apio/files/14368831/apio-errors.zip)

znmeb, 2024-02-22:
This looks like a Python "venv" problem. My install script installs apio in a virtual environment create by `python3 -m venv`. I tried it again with a `mamba` virtual environment and that one worked.

znmeb, 2024-02-22:
I found it - the directions are probably outdated. They're installing `apio` from a GitHub repo rather than from PyPI. If I just do `pip install --upgrade apio`, it works.

[apio-working.zip](https://github.com/FPGAwars/apio/files/14370484/apio-working.zip)

You can close this - I'll open an issue over on the `pico-ice` repository.

Obijuan, 2024-02-22:
They are installing the apio development version (because at that time their board was not supported in apio stable). But now it is supported in the stable version (from pypi)

The current development version has new changes that break some examples. I am currently adapting all the examples to this new version.

I will close this issue when i test all the examples

znmeb, 2024-02-22:
OK ... sounds good to me.

Obijuan, 2024-03-18:
@znmeb From apio 0.0.9 the top module should be included in the apio.ini file (the project file). By default the top module is `main`
In this example the top module is "rgb_test". 

In order to work ok, you should change this command:

`apio init --sayyes --board pico-ice `

by this one:

`apio init --sayyes --board pico-ice --top-module rgb_test`

(or change the top module of the verilog file from rgb_tst to main)

I will close this issue. It would be great to include  this example directly for the pico-ice board

In this link you can find the instructions on [how to add new examples in apio](https://github.com/FPGAwars/apio/wiki/The-apio-examples-package) . Pull request are very welcome

closed 2024-03-18 by Obijuan