FPGAwars/apio #351: [Feature request] Add a command to report the utilization info.

zapta, 2024-02-22:
This request is for a command similar to ``apio time`` that reports the FPGA utilization. Alternatively, the time and the utilization reporting can be combined into a single reporting command, that can be extended later to report additional information.

By utilization I mean something similar to this part of the the ``apio build -v`` log:

```
Info: Device utilisation:
Info: 	         ICESTORM_LC:    34/ 5280     0%
Info: 	        ICESTORM_RAM:     0/   30     0%
Info: 	               SB_IO:     4/   96     4%
Info: 	               SB_GB:     1/    8    12%
Info: 	        ICESTORM_PLL:     0/    1     0%
Info: 	         SB_WARMBOOT:     0/    1     0%
Info: 	        ICESTORM_DSP:     0/    8     0%
Info: 	      ICESTORM_HFOSC:     0/    1     0%
Info: 	      ICESTORM_LFOSC:     0/    1     0%
Info: 	              SB_I2C:     0/    2     0%
Info: 	              SB_SPI:     0/    2     0%
Info: 	              IO_I3C:     0/    2     0%
Info: 	         SB_LEDDA_IP:     0/    1     0%
Info: 	         SB_RGBA_DRV:     0/    1     0%
Info: 	      ICESTORM_SPRAM:     0/    4     0%
```

IceStudio shows this information in the status bar, scraped from this log test.  I wonder if the nextpnr team would agree to add a
``--report`` flag that emits the report as flag text or structured json.

obtitus, 2024-09-15:
don't know if it is new, but this information is now in .json when passing the `--report` argument to nextpnr:
```
$ nextpnr-ice40 --version
"nextpnr-ice40" -- Next Generation Place and Route (Version nextpnr-0.6-118-g0eb9a9ad)
$ nextpnr-ice40 --up5k --package sg48 --json hardware.json --asc hardware.asc --pcf pico_ice.pcf -q --report report.json

results in report.json:
...
"utilization": {                                                                                                                                                                                                                                      
    "ICESTORM_DSP": {                                                                                                                                                                                                                                   
      "available": 8,                                                                                                                                                                                                                                   
      "used": 0                                                                                                                                                                                                                                         
    },                                                                                                                                                                                                                                                  
    "ICESTORM_HFOSC": {                                                                                                                                                                                                                                 
      "available": 1,                                                                                                                                                                                                                                   
      "used": 0                                                                                                                                                                                                                                         
    },                                                                                                                                                                                                                                                  
    "ICESTORM_LC": {                                                                                                                                                                                                                                    
      "available": 5280,                                                                                                                                                                                                                                
      "used": 52                                                                                                                                                                                                                                        
    },
...
```

zapta, 2024-10-20:
Thanks @obtitus, this is very useful.

I am implementing a new command 'apio report' that prints the utilization and timing information from that file, replacing the 'apio command'.

zapta, 2024-11-12:
Implemented. Closing.

closed 2024-11-12 by zapta