FPGAwars/apio #358: [Proposal] Change Verible liner instead for Verilator.

zapta, 2024-02-28:
This is a proposal for discussion.

The idea is to switch the linter from Verilator to Veriable.

https://github.com/chipsalliance/verible

Rationale
In addition to a command line lint, Verible includes additional command line tools such as a verilog formatter or verilog aware diff, as well as a verilog language server that together with the the visual studio code verible plugin which provides verilog formatting and continuous linting while editing.

The actual code change to invoke verible instead of verilator is minimal. The main challange is to handle the automatic installation of the tools.

List of the Verible commands on my Mac OS:
```
verible-verilog-diff
verible-verilog-format
verible-verilog-kythe-extractor
verible-verilog-lint
verible-verilog-ls
verible-verilog-obfuscate
verible-verilog-preprocessor
verible-verilog-project
verible-verilog-syntax
```
@Obijuan , any thoughts?

Obijuan, 2024-03-26:
I agree with this propolsal, although  I have not used verible before
I've checked the upstream oss-cad-suite package, but it does not include verible
So my proposal is to include it directly in our oss-cad-suite apio package
It may take some time to be included in apio

Any help with this task is very welcome

zapta, 2024-12-22:
Per discussion in another issue here, we are going with verilator instead of verible.

The verible binaries are already installed now on all platform and we use it for the ``apio format`` command.

Closing.

closed 2024-12-22 by zapta