FPGAwars/apio #361: Have apio drivers --ftdi-enable printing debug information about zadig.

zapta, 2024-03-03:
I was trying to install apio on windows 10, from a git bash shell. When I run ``apio drivers --ftdi-enable`` the operation failed due to permissions, without giving a detailed information what it was tried to do.

This feature request is to add an info message that states what apio is trying to do such that the user can diagnose it. For example, the path to the zadig binary.

(The underlying issue is bash perimission here, I can run the zadig from a windows cmd shell but not from a bash sell, probably because I don't run it as an admin.)
```
$ apio drivers --ftdi-enable
(DEBUG) Profile path: C:\Users\user\.apio\profile.json
(DEBUG) Home_dir: C:\Users\user\.apio
Launch drivers configuration tool

   FTDI driver installation:
   Usage instructions

      1. Connect the FTDI FPGA board
      2. Select (Interface 0)
      3. Replace driver by "libusbK"
      4. Reconnect the board
      5. Check `apio system --lsftdi`

[WinError 740] The requested operation requires elevation
```

Obijuan, 2024-03-26:
Could you please test it in the apio development version? I've changed the error message. Check it and feel free to change it, modify or improve it. The code needs refactoring, but I will do it in the next cycle. Currently I am just fixing bugs for the 0.9.x stable release

Obijuan, 2024-03-26:
ok, i've also refactored the code (i could not resist it 😂️)

This is the new error message:

```
[WinError 740] The requested operation requires elevation
Trying to execute zadig.exe in command line, but an error ocurred
Please, execute the command again in the command line with administrator priviledges
```

As I am not a native English speaker, feel free to modify/change/improve the message

Obijuan, 2024-04-03:
Fixed in apio stable 0.9.3
I am closing this issue. Feel free to reopen it if the problem remains

closed 2024-04-03 by Obijuan