FPGAwars/apio #362: The progress of iceprog is not shown incrementally [Windows 10]

zapta, 2024-03-03:
On Mac OS, when I run 'apio upload' I see the progess of the operation with the iceprog messages that are printed in real time. On widnows 10, nothing is shown until the operation is end and then all the iceprog output is shown at once.

apio.ini
```
[env]
board = upduino31
top-module = main
```

Log
```
>apio upload
(DEBUG) Profile path: C:\Users\user\.apio\profile.json
(DEBUG) Home_dir: C:\Users\user\.apio
(DEBUG) Profile path: C:\Users\user\.apio\profile.json
(DEBUG) Home_dir: C:\Users\user\.apio
(DEBUG) Run Command: lsusb
(DEBUG) System_base_dir: C:\Users\user\.apio\packages\tools-oss-cad-suite
(DEBUG) System bin dir: C:\Users\user\.apio\packages\tools-oss-cad-suite\bin
(DEBUG) Executable file: C:\Users\user\.apio\packages\tools-oss-cad-suite\bin\lsusb.exe
(DEBUG) Profile path: C:\Users\user\.apio\profile.json
(DEBUG) Home_dir: C:\Users\user\.apio
(DEBUG) Run Command: lsftdi
(DEBUG) System_base_dir: C:\Users\user\.apio\packages\tools-oss-cad-suite
(DEBUG) System bin dir: C:\Users\user\.apio\packages\tools-oss-cad-suite\bin
(DEBUG) Executable file: C:\Users\user\.apio\packages\tools-oss-cad-suite\bin\lsftdi.exe

PATH: C:\Users\user\.apio\packages\tools-oss-cad-suite\bin;C:\Users\user\.apio\packages\tools-oss-cad-suite\lib;C:\Users\user\.apio\packages\tool-gtkwave\bin;C:\Users\user\.apio\packages\tool-gtkwave\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Common Files\Acronis\SnapAPI\;C:\Program Files (x86)\Common Files\Acronis\VirtualFile\;C:\Program Files (x86)\Common Files\Acronis\VirtualFile64\;C:\Program Files (x86)\Common Files\Acronis\FileProtector\;C:\Program Files (x86)\Common Files\Acronis\FileProtector64\;C;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Users\user\AppData\Local\Programs\Python\Python312\Scripts\;C:\Users\user\AppData\Local\Programs\Python\Python312\;C:\Users\user\AppData\Local\Microsoft\WindowsApps;C:\Users\user\AppData\Local\GitHubDesktop\bin;C:\Users\user\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files\Wireshark;C:\Users\user\AppData\Roaming\npm;

[Sun Mar  3 13:48:46 2024] Processing upduino31
------------------------------------------------------------------------------------------------------------------------
Warning: more than one testbenches used
iceprog -d i:0x0403:0x6014:0 hardware.bin
init..             <<-- everything from here is shown only when the operation ends.
cdone: high
reset..
cdone: low
flash ID: 0xEF 0x40 0x16 0x00
file size: 104090
erase 64kB sector at 0x000000..
erase 64kB sector at 0x010000..
programming..
reading..
VERIFY OK
cdone: high
Bye.
============================================= [SUCCESS] Took 15.57 seconds =============================================
```

Obijuan, 2024-03-26:
Could you please test it with the latest stable apio version (0.9.2)?. It should be fixed (the output in windows is not shown in the same way than in linux and mac, but now the output is refresh many times before ending)

Obijuan, 2024-04-03:
Latest stable version: 0.9.3
I think thus issue is solved. Please, feel free to reopen it if the problem remains

closed 2024-04-03 by Obijuan