FPGAwars/apio #363: apio graph fails when using tools-oss-cad-suite v0.0.9

gmsanchez, 2024-03-25:
I upgraded apio to version 0.9.2 and wanted to try the new ``apio graph`` command.

When I run it with tools-oss-cad-suite v0.0.8 the graphics render correctly. However, when I upgrade to tools-oss-cad-suite v0.0.9 I get this error:

```
$ apio graph
yosys -p "show -format dot -colors 1 -prefix hardware and_1_bit" -q and_1_bit.v
Warning: Selection "and_1_bit" did not match any module.
ERROR: Nothing there to show.
scons: *** [hardware.dot] Error 1
═══════════════════════════════════════════ [ ERROR ] Took 0.13 seconds ═══════════════════════════════════════════
```

The code I am trying to graph can be found here: [https://github.com/gmsanchez/edu-ciaa-fpga-verilog/blob/main/02-and/](https://github.com/gmsanchez/edu-ciaa-fpga-verilog/blob/main/02-and/)

Thanks!

gmsanchez, 2024-03-25:
When I run

```
$ apio raw 'yosys -p "show -format dot -colors 1 -prefix hardware and_1_bit" and_1_bit.v'

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.33+103 (git sha1 11ffd7df4, clang 10.0.0-4ubuntu1 -fPIC -Os)

-- Parsing `and_1_bit.v' using frontend ` -vlog2k' --

1. Executing Verilog-2005 frontend: and_1_bit.v
Parsing Verilog input from `and_1_bit.v' to AST representation.
Storing AST representation for module `$abstract\and_1_bit'.
Successfully finished Verilog frontend.

-- Running command `show -format dot -colors 1 -prefix hardware and_1_bit' --

2. Generating Graphviz representation of design.
Warning: Selection "and_1_bit" did not match any module.
Writing dot description to `hardware.dot'.
ERROR: Nothing there to show.
```

I notice that the frontend used is ``vlog2k``, while the frontend used in tools-oss-cad-suite v0.0.8 is ``verilog``. So I modified the command to the following

```
$ apio raw 'yosys -f verilog -p "show -format dot -colors 1 -prefix hardware and_1_bit" and_1_bit.v'

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.33+103 (git sha1 11ffd7df4, clang 10.0.0-4ubuntu1 -fPIC -Os)

-- Parsing `and_1_bit.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend: and_1_bit.v
Parsing Verilog input from `and_1_bit.v' to AST representation.
Generating RTLIL representation for module `\and_1_bit'.
Successfully finished Verilog frontend.

-- Running command `show -format dot -colors 1 -prefix hardware and_1_bit' --

2. Generating Graphviz representation of design.
Writing dot description to `hardware.dot'.
Dumping module and_1_bit to page 1.

End of script. Logfile hash: 74a7d6d8c2, CPU: user 0.00s system 0.01s, MEM: 9.50 MB peak
Yosys 0.33+103 (git sha1 11ffd7df4, clang 10.0.0-4ubuntu1 -fPIC -Os)
Time spent: 69% 1x show (0 sec), 30% 1x read_verilog (0 sec)
```

and I get the output file.

Obijuan, 2024-03-26:
thanks @gmsanchez !  I've added the -f verilog flag when using the "apio graph" command. You can try it with the apio development version (it will be soon releases in the stable apio 0.9.3 release)

commit 69650245 (2024-03-26): Bug fixed: #363: Error when executing apio graph
closed 2024-03-26 by Obijuan