FPGAwars/apio #364: Error when installing apio

zapta, 2024-04-04:
I am getting an error when trying to install apio 0.9.3 from scratch on mac osx.

It seems to be a broken reference to 

https://github.com/FPGAwars/tools-oss-cad-suite/releases/download/v0.0.9/tools-oss-cad-suite-darwin_arm64-0.0.9.tar.gz

The full log:
```
/Users/user/projects/fpga/repo/hdl $ python --version
Python 3.12.1
/Users/user/projects/fpga/repo/hdl $ pip freeze
/Users/user/projects/fpga/repo/hdl $ pip --version
pip 23.2.1 from /Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/pip (python 3.12)
/Users/user/projects/fpga/repo/hdl $ pip install apio --upgrade
Collecting apio
  Obtaining dependency information for apio from https://files.pythonhosted.org/packages/14/e1/eeb3a5b04fce7ec8d407b21b015ab5c75968c506bcb9e8d53c77d95b3e0f/apio-0.9.3-py3-none-any.whl.metadata
  Using cached apio-0.9.3-py3-none-any.whl.metadata (5.8 kB)
Collecting click==8.1.3 (from apio)
  Obtaining dependency information for click==8.1.3 from https://files.pythonhosted.org/packages/c2/f1/df59e28c642d583f7dacffb1e0965d0e00b218e0186d7858ac5233dce840/click-8.1.3-py3-none-any.whl.metadata
  Using cached click-8.1.3-py3-none-any.whl.metadata (3.2 kB)
Collecting semantic_version==2.9.0 (from apio)
  Obtaining dependency information for semantic_version==2.9.0 from https://files.pythonhosted.org/packages/64/ac/df31047966c4d0293e7bd16276ebc9f6654de36ad8e19061a09369380c0a/semantic_version-2.9.0-py2.py3-none-any.whl.metadata
  Using cached semantic_version-2.9.0-py2.py3-none-any.whl.metadata (9.7 kB)
Collecting requests==2.28.2 (from apio)
  Obtaining dependency information for requests==2.28.2 from https://files.pythonhosted.org/packages/d2/f4/274d1dbe96b41cf4e0efb70cbced278ffd61b5c7bb70338b62af94ccb25b/requests-2.28.2-py3-none-any.whl.metadata
  Using cached requests-2.28.2-py3-none-any.whl.metadata (4.6 kB)
Collecting colorama==0.4.6 (from apio)
  Obtaining dependency information for colorama==0.4.6 from https://files.pythonhosted.org/packages/d1/d6/3965ed04c63042e047cb6a3e6ed1a63a35087b6a609aa3a15ed8ac56c221/colorama-0.4.6-py2.py3-none-any.whl.metadata
  Using cached colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
Collecting pyserial==3.5 (from apio)
  Obtaining dependency information for pyserial==3.5 from https://files.pythonhosted.org/packages/07/bc/587a445451b253b285629263eb51c2d8e9bcea4fc97826266d186f96f558/pyserial-3.5-py2.py3-none-any.whl.metadata
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting wheel<1,>=0.35.0 (from apio)
  Obtaining dependency information for wheel<1,>=0.35.0 from https://files.pythonhosted.org/packages/7d/cd/d7460c9a869b16c3dd4e1e403cce337df165368c71d6af229a74699622ce/wheel-0.43.0-py3-none-any.whl.metadata
  Using cached wheel-0.43.0-py3-none-any.whl.metadata (2.2 kB)
Collecting scons==4.2.0 (from apio)
  Obtaining dependency information for scons==4.2.0 from https://files.pythonhosted.org/packages/d8/29/401703b6ce7ad8b4009ab16a435a7c08827357e1c7217ad39e750e75a358/SCons-4.2.0-py3-none-any.whl.metadata
  Using cached SCons-4.2.0-py3-none-any.whl.metadata (5.4 kB)
Collecting charset-normalizer<4,>=2 (from requests==2.28.2->apio)
  Obtaining dependency information for charset-normalizer<4,>=2 from https://files.pythonhosted.org/packages/3a/52/9f9d17c3b54dc238de384c4cb5a2ef0e27985b42a0e5cc8e8a31d918d48d/charset_normalizer-3.3.2-cp312-cp312-macosx_11_0_arm64.whl.metadata
  Using cached charset_normalizer-3.3.2-cp312-cp312-macosx_11_0_arm64.whl.metadata (33 kB)
Collecting idna<4,>=2.5 (from requests==2.28.2->apio)
  Obtaining dependency information for idna<4,>=2.5 from https://files.pythonhosted.org/packages/c2/e7/a82b05cf63a603df6e68d59ae6a68bf5064484a0718ea5033660af4b54a9/idna-3.6-py3-none-any.whl.metadata
  Using cached idna-3.6-py3-none-any.whl.metadata (9.9 kB)
Collecting urllib3<1.27,>=1.21.1 (from requests==2.28.2->apio)
  Obtaining dependency information for urllib3<1.27,>=1.21.1 from https://files.pythonhosted.org/packages/b0/53/aa91e163dcfd1e5b82d8a890ecf13314e3e149c05270cc644581f77f17fd/urllib3-1.26.18-py2.py3-none-any.whl.metadata
  Using cached urllib3-1.26.18-py2.py3-none-any.whl.metadata (48 kB)
Collecting certifi>=2017.4.17 (from requests==2.28.2->apio)
  Obtaining dependency information for certifi>=2017.4.17 from https://files.pythonhosted.org/packages/ba/06/a07f096c664aeb9f01624f858c3add0a4e913d6c96257acb4fce61e7de14/certifi-2024.2.2-py3-none-any.whl.metadata
  Using cached certifi-2024.2.2-py3-none-any.whl.metadata (2.2 kB)
Collecting setuptools (from scons==4.2.0->apio)
  Obtaining dependency information for setuptools from https://files.pythonhosted.org/packages/92/e1/1c8bb3420105e70bdf357d57dd5567202b4ef8d27f810e98bb962d950834/setuptools-69.2.0-py3-none-any.whl.metadata
  Using cached setuptools-69.2.0-py3-none-any.whl.metadata (6.3 kB)
Using cached apio-0.9.3-py3-none-any.whl (91 kB)
Using cached click-8.1.3-py3-none-any.whl (96 kB)
Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Using cached requests-2.28.2-py3-none-any.whl (62 kB)
Using cached SCons-4.2.0-py3-none-any.whl (4.2 MB)
Using cached semantic_version-2.9.0-py2.py3-none-any.whl (15 kB)
Using cached wheel-0.43.0-py3-none-any.whl (65 kB)
Using cached certifi-2024.2.2-py3-none-any.whl (163 kB)
Using cached charset_normalizer-3.3.2-cp312-cp312-macosx_11_0_arm64.whl (119 kB)
Using cached idna-3.6-py3-none-any.whl (61 kB)
Using cached urllib3-1.26.18-py2.py3-none-any.whl (143 kB)
Using cached setuptools-69.2.0-py3-none-any.whl (821 kB)
Installing collected packages: pyserial, wheel, urllib3, setuptools, semantic_version, idna, colorama, click, charset-normalizer, certifi, scons, requests, apio
Successfully installed apio-0.9.3 certifi-2024.2.2 charset-normalizer-3.3.2 click-8.1.3 colorama-0.4.6 idna-3.6 pyserial-3.5 requests-2.28.2 scons-4.2.0 semantic_version-2.9.0 setuptools-69.2.0 urllib3-1.26.18 wheel-0.43.0
WARNING: There was an error checking the latest version of pip.
/Users/user/projects/fpga/repo/hdl $ apio install -a
File version.txt downloaded!
Version: 0.0.36
Installing examples package:
Download apio-examples-0.0.36.zip
Downloading  [████████████████████████████████████]  100%
Unpacking..  [████████████████████████████████████]  100%
Package 'examples' has been successfully installed!
File version.txt downloaded!
Version: 0.0.9
Installing oss-cad-suite package:
Got an unrecognized status code: 404
When downloading https://github.com/FPGAwars/tools-oss-cad-suite/releases/download/v0.0.9/tools-oss-cad-suite-darwin_arm64-0.0.9.tar.gz
Error: Package not found
```

zapta, 2024-04-04:
Is the right link should be this?

https://github.com/FPGAwars/tools-oss-cad-suite/releases/download/v0.1.0/tools-oss-cad-suite-darwin_arm64-0.1.0.tar.gz

zapta, 2024-04-04:
> Yes, It is a bug... apio 0.9.3 comes by default with oss-cad-suite-0.0.9
> (the version is correct), but not the generated name for the package
> You can quick fix it just by typing: apio install ***@***.***,
> meanwhile I will try to find the bug
> In order to get more information about it I need you execute this command
> and show me the output:
>
> apio system --info

Thanks. The install command you gave me worked.  Here is the system info

```

$ apio system --info

Platform: darwin_arm64

```

Obijuan, 2024-04-04:
Thanks for the information. It is fixed. I will close the issue (Please, reopen it if the problem remains)

closed 2024-04-04 by Obijuan

zapta, 2024-04-04:
I installed apio from scratch and it was able to install tools-oss-cad-suite-darwin_arm64-0.0.9.tar.gz with no problem. Thanks!