FPGAwars/apio #367: Please add a doc entry for apio.ini

zapta, 2024-04-04:
Please add to APIO's documentation of page for documenting apio.ini, similar to the one for platformio.ini https://docs.platformio.org/en/latest/projectconf/

I plan to work on thing that will more apio.ini a more prominent feature of apio (e.g. https://github.com/FPGAwars/apio/issues/357) and being able to refer users to a canonical documentation page will be very useful.

At this stage, even a skeleton with a stable URL will be useful.

Obijuan, 2024-04-04:
Hi!  The latest apio documentation is on the wiki. I've updated it recently. You can find that information in this link:
https://github.com/FPGAwars/apio/wiki/Project-configuration-file

I think that you, as an FPGAwar member, have permissions to edit the wiki page (please, check it).  If not, tell me and I will give them to you, so that you can edit, add or modify the documentation as you like

zapta, 2024-04-04:
What is the status of the documentation here https://apiodoc.readthedocs.io/en/stable/index.html ?  It looks nice. Is it still relevant? Where are the source files?

Obijuan, 2024-04-05:
> What is the status of the documentation here https://apiodoc.readthedocs.io/en/stable/index.html ?  

It is obsolete. No longer used (since apio 0.0.4). The doc was moved to the wiki page

>  It looks nice. Is it still relevant? Where are the source files?

The source files were in the docs folder in the apio repo, but I've just removed them in apio 0.9.3

> EDIT: I updated https://github.com/FPGAwars/apio/wiki/Project-configuration-file with the current attributes.

Great! thanks! :smiley:

> Can you fill in the description of ext-mode ? I am not sure what's the difference between 'native' and 'default'.

Yes, for sure!

zapta, 2024-04-06:
Since we have a doc page for apio.ini, closing this one.

closed 2024-04-06 by zapta