FPGAwars/apio #369: windows go board apio upload fails

pm100, 2024-04-04:
I followed the install instructions and got to apio upload

```
PS C:\work\nandland\go-board\Blinky> apio upload
[Thu Apr  4 15:47:33 2024] Processing go-board
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
Command not found:
(['scons', '-Q', 'upload', 'fpga_arch=ice40', 'fpga_size=1k', 'fpga_type=hx', 'fpga_pack=vq100', 'top_module=Test', 'prog=iceprog -d i:0x0403:0x6010:0', '-f', 'C:\\Python310\\lib\\site-packages\\apio\\resources\\ice40\\SConstruct'],)
```

Ran as normal user and admin, same result

zapta, 2024-04-05:
IIt seems that the 'scons' command is not found.  

1. What happens when you run ```scons``` from the command line?

2. When you run ```pip freeze```, do you see a line with 'scons' ?

pm100, 2024-04-05:
```
PS C:\work\nandland\go-board\Blinky> scons
scons : The term 'scons' is not recognized as the name of a cmdlet, function, script file, or operable program. Check
the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ scons
+ ~~~~~
    + CategoryInfo          : ObjectNotFound: (scons:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

PS C:\work\nandland\go-board\Blinky> pip freeze
WARNING: Ignoring invalid distribution -ip (c:\python310\lib\site-packages)
antlr4-tools==0.2.1
apio==0.9.3
aubio==0.4.9
certifi==2024.2.2
charset-normalizer==3.3.2
click==8.1.3
colorama==0.4.6
idna==3.6
install-jdk==1.1.0
numpy==1.23.4
PyAudio==0.2.12
pyserial==3.5
requests==2.28.2
SCons==4.2.0
semantic-version==2.9.0
urllib3==1.26.18
PS C:\work\nandland\go-board\Blinky>
```

should I have an executable called scons somewhere, I do not have one anywhere

zapta, 2024-04-05:
... this is what I see in that python bin directory

```
/Library/Frameworks/Python.framework/Versions/3.12/bin $ ls -ald sco*
-rwxr-xr-x  1 user  admin  222 Apr  4 19:22 scons
-rwxr-xr-x  1 user  admin  235 Apr  4 19:22 scons-configure-cache
-rwxr-xr-x  1 user  admin  229 Apr  4 19:22 sconsign
```

And the scons file contains this
```
/Library/Frameworks/Python.framework/Versions/3.12/bin $ cat scons
#!/usr/local/bin/python3
# -*- coding: utf-8 -*-
import re
import sys
from SCons.Script.Main import main
if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw|\.exe)?$', '', sys.argv[0])
    sys.exit(main())
```

zapta, 2024-04-05:
... looking at my shell config file .bashrc, I see that I manually added that directory at some point to my path.  You may need to do something similar on your windows machine. This seems to be standard python stuff so you may want to look around.

```
export PATH="$PATH:/Library/Frameworks/Python.framework/Versions/3.12/bin"
```

zapta, 2024-04-05:
... I checked on my windows machine. The scons file is here

```
C:\Users\user>dir c:\Users\user\AppData\Local\Programs\Python\Python312\Scripts\scons.exe
 Volume in drive C has no label.
 Volume Serial Number is 0EAC-4058

 Directory of c:\Users\user\AppData\Local\Programs\Python\Python312\Scripts

03/03/2024  02:15 PM           108,418 scons.exe
               1 File(s)        108,418 bytes
               0 Dir(s)  85,971,275,776 bytes free
```

pm100, 2024-04-05:
so I was supposed to install it myself. The apio installation instruction dont mention it at all

Obijuan, 2024-04-05:
Scons is installed automatically with apio. You do not have to install by your own. But for unkown reason it has not been installed correctly in your system, or it is not in your Path

Try to install it manually. Run this command:

`pip install scons`

It would be great if you could post the output. Maybe, readi.g it,  we can get more information about the problem

zapta, 2024-04-05:
Since `SCons==4.2.0` already appears in your `pip freeze` output, my guess is that you need to add the Scripts directory to your path. The bug report below suggests that this is because you didn't check "Add Python to PATH" when you installed python.

https://github.com/python/cpython/issues/111434

pm100, 2024-04-05:
I found I had another msys/mingw package installed (Strawberry) its dlls were interfering with apio/ypsys/scons etc.

Removed it, uninstalled apio, python,...... reinstalled from scratch

all works now, but see #370 - apio uninstall does not work on windows

closed 2024-04-05 by pm100