FPGAwars/apio #370: apio uninstall does not work on windows

pm100, 2024-04-05:
As part of me trying to sort out my scons issues I decided to reinstall apio etc.

apio uninstall -a fails

```
PS C:\Users\paulm> apio uninstall -a
Do you want to continue? [y/N]: y
Uninstalling gtkwave package:
Traceback (most recent call last):
  File "C:\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Python310\Scripts\apio.exe\__main__.py", line 7, in <module>
  File "C:\Python310\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Python310\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Python310\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Python310\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Python310\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Python310\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Python310\lib\site-packages\apio\commands\uninstall.py", line 81, in cli
    _uninstall(packages, platform)
  File "C:\Python310\lib\site-packages\apio\commands\uninstall.py", line 38, in _uninstall
    inst.uninstall()
  File "C:\Python310\lib\site-packages\apio\managers\installer.py", line 351, in uninstall
    shutil.rmtree(file)
  File "C:\Python310\lib\shutil.py", line 739, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Python310\lib\shutil.py", line 612, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python310\lib\shutil.py", line 612, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python310\lib\shutil.py", line 612, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python310\lib\shutil.py", line 617, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Python310\lib\shutil.py", line 615, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\paulm\\.apio\\packages\\tool-gtkwave\\lib\\tcl8.6\\msgs\\af_za.msg'
PS C:\Users\paulm>
```

Using procmon i see that the code is using SetDispositionInformationEx and that fails with CANNOT DELETE because the fille was marked 'read only ' when installed and the docs say

>A return value of STATUS_CANNOT_DELETE indicates that either the file is read-only, or there is an existing mapped view to the file.

If I change the file to be read-write then that file gets deleted an a rerun, but then the next one fails.

Yes I am running as administrator

pm100, 2024-04-05:
changed all file to be read write in file explorer - uninstalled ok

zapta, 2024-11-12:
Can not reproduce. Is it possible that the gtkwave tool was opened from another shell window?

The gtkwave package is going away (using gtkwave from oss-cad-suite package) so closing this issue.

closed 2024-11-12 by zapta