FPGAwars/apio #371: apio graph fails - cant find 'dot'

pm100, 2024-04-05:
Windows, fresh install of everything; 'apio upload ' works fine

```
PS C:\work\nandland\go-board\Blinky> apio graph
yosys -f verilog -p "show -format dot -colors 1 -prefix hardware Test" -q Blinky.v
dot -Tsvg hardware.dot -o hardware.svg
'dot' is not recognized as an internal or external command,
operable program or batch file.
scons: *** [hardware.svg] Error 1
═════════════════════════════════════════════ [ ERROR ] Took 0.36 seconds ═════════════════════════════════════════════
PS C:\work\nandland\go-board\Blinky>

```

zapta, 2024-04-06:
Thanks @pm100.  'graph' is a new command that was added recently. As a woraround, you can install the graphviz package manually as explain at https://graphviz.org/download/ . This should provide the 'dot' command.

@Obijuan, the dot command used to convert the .dot file that yosys generates to .svg. Any idea how to have it installed automatically? 

```
yosys -f verilog -p "show -format dot -colors 1 -prefix hardware main" -q fifo_tx.v i2s_rx.v i2s_test_pattern.v i2s_timing.v main.v meta_stability.v queue.v queue_pusher.v register.v reset_gen.v sine_gen.v status_leds.v

dot -Tsvg hardware.dot -o hardware.svg
```

pm100, 2024-04-09:
yes , installing grpahviz solved my problem, Maybe just a doc update needed

zapta, 2024-10-13:
The next release has this in the output of  ``apio graph -h``:

```
The dot command is not included with the apio distribution and needed to be
installed seperatly. See https://graphviz.org for more details.
```

Closing.

closed 2024-10-13 by zapta