FPGAwars/apio #373: Where can i fnd and use the "icesprog" program?

Nov4ou, 2024-04-10:
I currently using a M1 MacBook Air and iCESugar 1.5 FPGA board I had installed the icestrom toolchain and I find out I can't use the "iceprog", and should use""icesprog" instead. I tried to build the icesprog tool from source code but failed.

Fortunately，I can use apio to upload .bin to my board. But I want to learn to use Makefile to build my project then upload to my board by my own. So could you tell me where is the iceporg integrated in apio, or help me how  to build  the tool?
Thanks!

zapta, 2024-04-18:
On my Mac it's in the path below.

```
$ which iceprog
/Users/user/.apio/packages/tools-oss-cad-suite/bin/iceprog
```

Did you run the command to install all packages? If not, try this:

```
apio install --all
```

zapta, 2024-10-15:
Clsoing this issue. Please feel free to open if there are still open points.

closed 2024-10-15 by zapta