FPGAwars/apio #375: Feature Request: Support generic FTDI boards for programming

jasonsbeer, 2024-04-20:
I have an Adafruit FT232H board I wish to use for programming SPI flash memory for the iCE40HX4K. This is one example of many generic FTDI boards out there. My intent is to program the flash memory in a project. I am not using a dev board. Reviewing the documentation for the UPLOAD command, there does not appear to be any way to point to a generic FTDI board, such as this.

My apologies if this capability already exists, but I can find no documented procedures. Please see the output below. I am using Ubuntu 23.10.

`(.venv) jason@lab-computer:~/Documents/GitHub/AmigaPCI/Logic/U712$ apio system --lsftdi`
`ftdi_usb_get_strings failed: -7 (libusb_get_string_descriptor_ascii() failed)`
`Number of FTDI devices found: 1`
`Checking device: 0`

Documentaion cited: https://github.com/FPGAwars/apio/wiki/Apio-Upload

Thanks for this project and all your hard work.

jasonsbeer, 2024-04-20:
I just needed to dig a little further to see I can use ICEPROG to accomplish this task. Hopefully this will help someone in the future. Please close.

zapta, 2024-10-15:
@jasonbeer, apio works by board id, which allows it to retrieve information such as the FPGA used, the programmer to use, special arguments, and so on.

If you want to program a custom iCE40-HX4K board using FT232H, you can use a board with similar FPGA and a similar FTDI ic and connectivity and pretend that this is your board. You may need to program your FT232H to have the proper ID and configuration (using the FT PROG utility from FTDI). For example, this board https://github.com/FPGAwars/icezum/blob/master/doc/schematics/icezum.pdf .

In the next  release of apio, you will also be able to have in your project a custom definition of your board (similar to the stock boards.json of apio  https://github.com/FPGAwars/apio/blob/develop/apio/resources/boards.json ).

Hope that this address your question. I will close this issue now but please feel free to reopen or create a new one if there are still open points.

closed 2024-10-15 by zapta