FPGAwars/apio #376: ERROR: Module `main' not found!

yoo142, 2024-04-26:
I recieve the following error when trying to build or upload a projectusing APIO :  "ERROR: Module `main' not found! scons: *** [hardware.json] Error 1"
[image]

yoo142, 2024-04-26:
Apologies, I just figured it out, I didn't realize the top module actually had to be named "main". Is this a new feature or am I going crazy? I swear I have built projects without having the module named "main".

Obijuan, 2024-04-26:
Please, read the [release notes](https://github.com/FPGAwars/apio/wiki/Release-history#version-091)
As the toolchains has been updated, now the top module should be explicitely given in the apio.ini project file
If it is not given, apio will use main as the default
You can add the module name manually in the apio.ini file, or you can use the init command:
`apio init -t <yout-top-module-name>`

You can also find more information about the init command in the wiki: [Init command](https://github.com/FPGAwars/apio/wiki/Apio-init)

closed 2024-06-12 by Obijuan