FPGAwars/apio #378: apio sim gives `Error: module 'click' has no attribute 'get_terminal_size'`

qcsui, 2024-05-02:
I think the reference to click is outdated.

> click.get_terminal_size() is deprecated and will be removed in 8.1. Use shutil.get_terminal_size instead. 1736
https://github.com/pallets/click/blob/2785712a8cf10820355236201dbb7751e917442f/CHANGES.rst#L142

zapta, 2024-09-04:
Seems to be fixed. All references to get_terminal_size() I could find use shutil. Closing.

closed 2024-09-05 by Obijuan

VA3IQ, 2024-10-24:
Just completed an apio install.  Everything went well until I tried a "apio verify".  I am getting the  Error: module 'click' has no attribute 'get_terminal_size'

zapta, 2024-10-24:
@VA3IQ, do you get a stack trace with the error? If so, can you post it here?

I installed the release apio version (0.9.5) and can't find references to ``click.get_terminal_size``.

```
/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/apio $ grep -r get_terminal_size *
Binary file __pycache__/resources.cpython-312.pyc matches
Binary file managers/__pycache__/examples.cpython-312.pyc matches
Binary file managers/__pycache__/scons.cpython-312.pyc matches
managers/scons.py:        terminal_width, _ = shutil.get_terminal_size()
managers/examples.py:        terminal_width, _ = shutil.get_terminal_size()
resources.py:        terminal_width, _ = shutil.get_terminal_size()
resources.py:        terminal_width, _ = shutil.get_terminal_size()
resources.py:        terminal_width, _ = shutil.get_terminal_size()
```

VA3IQ, 2024-10-24:
There was no stack trace. :-(.

I forced an upgrade on the install (python -m pip install app -U),
confirmed driver, and created a new ini for the project.  Success!  Perhaps
the problem was between the chair and keyboard but all is working now.

Your help is much appreciated.

zapta, 2024-10-24:
Very good. Thanks for the update.