FPGAwars/apio #379: apio 0.94 fails to install oss-cad-suite on x86_64 mac because the link is broken

adennen, 2024-05-10:
It looks like apio is following an invalid link when trying to download oss-cad-suite on macos

this link:
https://github.com/FPGAwars/tools-oss-cad-suite/releases/download/v0.0.9/tools-oss-cad-suite-darwin_x86_64-0.0.9.tar.gz

should be changed to:
https://github.com/FPGAwars/tools-oss-cad-suite/releases/download/v0.0.9/tools-oss-cad-suite-darwin-0.0.9.tar.gz

```
apio install --all
File version.txt downloaded!
Version: 0.0.36
Installing examples package:
Already installed. Version 0.0.36
File version.txt downloaded!
Version: 0.0.9
Installing oss-cad-suite package:
Got an unrecognized status code: 404
When downloading https://github.com/FPGAwars/tools-oss-cad-suite/releases/download/v0.0.9/tools-oss-cad-suite-darwin_x86_64-0.0.9.tar.gz
Error: Package not found
```

Obijuan, 2024-06-12:
Hi @adennen !

Could you please show me the output of this command when executed in your system?
```
apio system -i
```

@zapta , I think you are also using apio in Mac, could you please check what happens in your system when execute the `apio install --all`. Can you reproduce the bug? (It would also be great if you could provide the output of the command `apio system -l` in your mac. Thanks a lot

zapta, 2024-06-12:
Juan, the information is below. Please let me know if you need anything else. `apio system` -l gave me an error (no such flag -l, so I skipped it). My computer is Mac with Apple M2 silicon (non x86).

Script:
```
#!/bin/bash -x

set -e

# Sys info
system_profiler SPSoftwareDataType SPHardwareDataType

pip freeze | xargs pip uninstall -y

rm -rf ~/.apio

pip freeze

pip install apio

pip install packaging

which apio

#apio system -l

apio install --all

apio clean

apio build

```

Log:
```
/Users/user/projects/fpga/repo/hdl $ ./test.sh 
+ set -e
+ system_profiler SPSoftwareDataType SPHardwareDataType
Software:

    System Software Overview:

      System Version: macOS 14.5 (23F79)
      Kernel Version: Darwin 23.5.0
      Boot Volume: Macintosh HD
      Boot Mode: Normal
      Computer Name: User’s MacBook Pro
      User Name: User (user)
      Secure Virtual Memory: Enabled
      System Integrity Protection: Enabled
      Time since boot: 2 days, 4 hours, 19 minutes

Hardware:

    Hardware Overview:

      Model Name: MacBook Pro
      Model Identifier: Mac14,7
      Chip: Apple M2
      Total Number of Cores: 8 (4 performance and 4 efficiency)
      Memory: 8 GB
      System Firmware Version: 10151.121.1
      OS Loader Version: 10151.121.1

+ pip freeze
+ xargs pip uninstall -y
Found existing installation: apio 0.9.4
Uninstalling apio-0.9.4:
  Successfully uninstalled apio-0.9.4
Found existing installation: certifi 2024.6.2
[... 96 lines cut ...]
Unpacking..  [████████████████████████████████████]  100%
Package 'examples' has been successfully installed!
File version.txt downloaded!
Version: 0.0.9
Installing oss-cad-suite package:
Download tools-oss-cad-suite-darwin_arm64-0.0.9.tar.gz
Downloading  [████████████████████████████████████]  100%          
Unpacking..  [████████████████████████████████████]  100%          
Package 'oss-cad-suite' has been successfully installed!
+ apio clean
Removed hardware.json
Removed hardware.asc
Removed hardware.bin
======================================================= [SUCCESS] Took 0.08 seconds =======================================================
+ apio build
[Wed Jun 12 13:49:41 2024] Processing upduino31
-------------------------------------------------------------------------------------------------------------------------------------------
yosys -p "synth_ice40 -top main -json hardware.json" -q fifo_tx.v i2s_rx.v i2s_test_pattern.v i2s_timing.v main.v meta_stability.v queue.v queue_pusher.v register.v reset_gen.v status_leds.v
Warning: Replacing memory \arr with list of registers. See i2s_rx.v:34
nextpnr-ice40 --up5k --package sg48 --json hardware.json --asc hardware.asc --pcf main.pcf -q
icepack hardware.asc hardware.bin
======================================================= [SUCCESS] Took 5.95 seconds =======================================================
/Users/user/projects/fpga/repo/hdl $ 

```

Obijuan, 2024-06-13:
> Juan, the information is below. Please let me know if you need anything else. apio system -l gave me an error (no such flag -l, so I skipped it). My computer is Mac with Apple M2 silicon (non x86).

ups, i made a miskate. I want you to test this command instead: ```apio system -i```

@adennen I also need you to test this command and show me the output, in order to fix this bug: ```apio system -i```
I will try to fix it for the next bug release (apio 0.9.5) in a few days

zapta, 2024-06-13:
$ apio system -i

Platform: darwin_arm64

Obijuan, 2024-06-14:
> $ apio system -i
Platform: darwin_arm64

Thanks for the test! There is no bug for the darwin_arm64 platform. I need @adennen to provide information about its system in order to fix the bug. If not provided I will release apio 0.9.5 stable. If the bug is confirmed we can fix it in the next release cycle

Obijuan, 2024-06-14:
I think I've fixed it in apio stable 0.9.5. I still need @adennen to execute `apio system -i` for getting more information about the bug  
I also need @adennen to install apio 0.9.5 and execute `apio install --all` for confirming that the bug is fixed
I will keep this issue open until I got some feedback

Obijuan, 2024-06-21:
I am closing this issue as there are no feedback. Please, feel free to reopen it if the bug remains

closed 2024-06-21 by Obijuan