FPGAwars/apio #380: Missing package 'packaging' in APIO installation

zapta, 2024-06-02:
When installing apio from scratch on Mac OSX, the package 'package is missing'.

I reproduce it with this shell script that **deletes all pip packages** and then install apio and runs it with no args.  Installing the package 'packaging' manually solves the problem.

```
#!/bin/bash -x

python --version
pip freeze | xargs pip uninstall -y
pip freeze
pip install apio
apio
```

Output log. Error message is at the end.

```
/Users/user/projects/fpga/repo/hdl $ ./test.sh 
+ python --version
Python 3.12.1
+ pip freeze
+ xargs pip uninstall -y
Found existing installation: apio 0.9.4
Uninstalling apio-0.9.4:
  Successfully uninstalled apio-0.9.4
Found existing installation: certifi 2024.6.2
Uninstalling certifi-2024.6.2:
  Successfully uninstalled certifi-2024.6.2
Found existing installation: charset-normalizer 3.3.2
Uninstalling charset-normalizer-3.3.2:
  Successfully uninstalled charset-normalizer-3.3.2
Found existing installation: click 8.1.3
Uninstalling click-8.1.3:
  Successfully uninstalled click-8.1.3
Found existing installation: colorama 0.4.6
Uninstalling colorama-0.4.6:
  Successfully uninstalled colorama-0.4.6
Found existing installation: idna 3.7
Uninstalling idna-3.7:
  Successfully uninstalled idna-3.7
Found existing installation: pyserial 3.5
Uninstalling pyserial-3.5:
  Successfully uninstalled pyserial-3.5
Found existing installation: requests 2.28.2
Uninstalling requests-2.28.2:
  Successfully uninstalled requests-2.28.2
Found existing installation: SCons 4.2.0
Uninstalling SCons-4.2.0:
  Successfully uninstalled SCons-4.2.0
Found existing installation: semantic-version 2.9.0
Uninstalling semantic-version-2.9.0:
  Successfully uninstalled semantic-version-2.9.0
Found existing installation: setuptools 70.0.0
Uninstalling setuptools-70.0.0:
  Successfully uninstalled setuptools-70.0.0
Found existing installation: urllib3 1.26.18
Uninstalling urllib3-1.26.18:
  Successfully uninstalled urllib3-1.26.18
Found existing installation: wheel 0.43.0
Uninstalling wheel-0.43.0:
  Successfully uninstalled wheel-0.43.0
+ pip freeze
+ pip install apio
Collecting apio
  Using cached apio-0.9.4-py3-none-any.whl.metadata (5.8 kB)
Collecting click==8.1.3 (from apio)
  Using cached click-8.1.3-py3-none-any.whl.metadata (3.2 kB)
Collecting semantic_version==2.9.0 (from apio)
  Using cached semantic_version-2.9.0-py2.py3-none-any.whl.metadata (9.7 kB)
Collecting requests==2.28.2 (from apio)
  Using cached requests-2.28.2-py3-none-any.whl.metadata (4.6 kB)
Collecting colorama==0.4.6 (from apio)
  Using cached colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
Collecting pyserial==3.5 (from apio)
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting wheel<1,>=0.35.0 (from apio)
  Using cached wheel-0.43.0-py3-none-any.whl.metadata (2.2 kB)
Collecting scons==4.2.0 (from apio)
  Using cached SCons-4.2.0-py3-none-any.whl.metadata (5.4 kB)
Collecting charset-normalizer<4,>=2 (from requests==2.28.2->apio)
  Using cached charset_normalizer-3.3.2-cp312-cp312-macosx_11_0_arm64.whl.metadata (33 kB)
Collecting idna<4,>=2.5 (from requests==2.28.2->apio)
  Using cached idna-3.7-py3-none-any.whl.metadata (9.9 kB)
Collecting urllib3<1.27,>=1.21.1 (from requests==2.28.2->apio)
  Using cached urllib3-1.26.18-py2.py3-none-any.whl.metadata (48 kB)
Collecting certifi>=2017.4.17 (from requests==2.28.2->apio)
  Using cached certifi-2024.6.2-py3-none-any.whl.metadata (2.2 kB)
Collecting setuptools (from scons==4.2.0->apio)
  Using cached setuptools-70.0.0-py3-none-any.whl.metadata (5.9 kB)
Using cached apio-0.9.4-py3-none-any.whl (91 kB)
Using cached click-8.1.3-py3-none-any.whl (96 kB)
Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Using cached requests-2.28.2-py3-none-any.whl (62 kB)
Using cached SCons-4.2.0-py3-none-any.whl (4.2 MB)
Using cached semantic_version-2.9.0-py2.py3-none-any.whl (15 kB)
Using cached wheel-0.43.0-py3-none-any.whl (65 kB)
Using cached certifi-2024.6.2-py3-none-any.whl (164 kB)
Using cached charset_normalizer-3.3.2-cp312-cp312-macosx_11_0_arm64.whl (119 kB)
Using cached idna-3.7-py3-none-any.whl (66 kB)
Using cached urllib3-1.26.18-py2.py3-none-any.whl (143 kB)
Using cached setuptools-70.0.0-py3-none-any.whl (863 kB)
Installing collected packages: pyserial, wheel, urllib3, setuptools, semantic_version, idna, colorama, click, charset-normalizer, certifi, scons, requests, apio
Successfully installed apio-0.9.4 certifi-2024.6.2 charset-normalizer-3.3.2 click-8.1.3 colorama-0.4.6 idna-3.7 pyserial-3.5 requests-2.28.2 scons-4.2.0 semantic_version-2.9.0 setuptools-70.0.0 urllib3-1.26.18 wheel-0.43.0
+ apio
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.12/bin/apio", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/click/core.py", line 1635, in invoke
    rv = super().invoke(ctx)
         ^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/apio/__main__.py", line 123, in cli
    _help = ctx.get_help()
            ^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/click/core.py", line 699, in get_help
    return self.command.get_help(self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/click/core.py", line 1298, in get_help
    self.format_help(ctx, formatter)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/click/core.py", line 1331, in format_help
    self.format_options(ctx, formatter)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/click/core.py", line 1533, in format_options
    self.format_commands(ctx, formatter)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/click/core.py", line 1586, in format_commands
    cmd = self.get_command(ctx, subcommand)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/apio/__main__.py", line 74, in get_command
    eval(code, nnss, nnss)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/apio/commands/upgrade.py", line 12, in <module>
    from packaging import version
ModuleNotFoundError: No module named 'packaging'
/Users/user/projects/fpga/repo/hdl $ 

```

Obijuan, 2024-06-12:
You can test it in the master branch, or you can wait some days when it is released as apio 0.9.5. I will leave this issue open util you can test it and confirm that is has been fixed

zapta, 2024-06-12:
Thanks for fixing it. I will test on the external release. Please ping me if I will forget.

Obijuan, 2024-06-14:
@zapta, I've just released apio 0.9.5. I need you to confirm if this bug has been fixed or not. Thanks
I will keep this issue open until I have some feedback

zapta, 2024-06-14:
I just tested 0.0.5 and it works. Thanks for fixing it.

Obijuan, 2024-06-14:
Thanks! I am closing this issue then

closed 2024-06-14 by Obijuan