FPGAwars/apio #381: Support Xilinix boards

Kreijstal, 2024-06-07:
I would like for apio to support the Xilinix FPGA, does that make sense?

zapta, 2025-05-07:
@cavearr, can you give me access to issue label editing?  I would like to add a 'architecture request' and 'language request' labels.

cavearr, 2025-05-07:
Hi @zapta ! we can't granularize permissions and your role is under "admin" .I'm creating this two labels for you. Thanks and sorry for the inconvenience.

zapta, 2025-05-07:
Thanks @cavearr.

zapta, 2025-05-07:
@cavearr, can you add two new label 'cleanup' and 'process' ?

cavearr, 2025-05-07:
Here you are! @zapta

zapta, 2026-05-24:
Thanks Carlos, I will be happy to test it.

cavearr, 2026-09-16:
@Kreijstal 

The condition @Obijuan set for closing this is met: #890 (Windows) and #891 (macOS) closed in June, and since August the openxc7 package ships for the three apio platforms (linux-x86-64, darwin-arm64, windows-amd64), with the per-device database downloaded on demand so the package stays around 100 MB. 

The definitions today carry 23 Xilinx boards across artix7, spartan7 and zynq7 — Arty A7 35T/100T, Arty S7 25/50, Basys3, Nexys A7, Zybo Z7 10/20, QMTech, Alinx, Alchitry Au and more — and apio 1.6.0 (#1037) is the first PyPI release to include all of that. 

Proposing to close; new board requests can be their own issues against fpgawars/apio-definitions.

cc: @zapta

closed 2026-09-16 by cavearr

Obijuan, 2026-09-16:
Amazing! Thanks for all you work