FPGAwars/apio #385: [Feature Request] Provide an apio plugin for Visual Studio Code.

zapta, 2024-06-21:
Visual Studio Code is a very popular general purpose IDE if not the most popular. This request is for a VSC plugin for apio that will simplified the installation and use of APIO. The requirements are straight forward and platformio's successful plugin can be used as a good starting point.

Key requirements
1. Automatic and software installation and updates are done by the plugin. 
2. The plugin is isolated from the system, having its own python environment and other necessary dependencies.
3. Allows to edit/verify/build/test/simulate/upload
4. Verilog editing support such as coloring and code formatting.
5. Creation of skeleton new project (user selects board).

zapta, 2025-11-09:
Hi @cavearr, can you create a `apio-vscode` repository and give me access? 

Please use the same configuration as apio-examples (issues disabled, workflow action enabled, etc)

zapta, 2025-11-10:
Hi @cavearr, can you create the repository?

cavearr, 2025-11-10:
Done! sorry for the delay! @zapta

zapta, 2025-11-11:
Thanks @cavearr !

cavearr, 2025-11-15:
Very interesting feature @zapta ! i'm deal with it this weekend i'm telling you soon.