FPGAwars/apio #388: apio lint fails to find verilated_std.sv

obtitus, 2024-07-07:
Hi, installed on Ubuntu 22.04.4 LTS
```
$ python3 --version
Python 3.10.12
$ python3 -m venv venv;source venv/bin/activate;pip install apio
$ apio --version
apio, version 0.9.5
$ apio system --info
Platform: linux_x86_64
$ apio install -a
$ sudo apt-get install iverilog graphviz
$ cd <and/project/directory>
$ apio lint
verilator --lint-only --timing -Wno-TIMESCALEMOD And.v And_top.v And_tb.v
%Error: Cannot find verilated_std.sv containing built-in std:: definitions:/yosyshq/share/verilator/include/verilated_std.sv
%Error: This may be because there's no search path specified with -I<dir>.
... Looked in: 
/yosyshq/share/verilator/include/verilated_std.sv
/yosyshq/share/verilator/include/verilated_std.sv.v
/yosyshq/share/verilator/include/verilated_std.sv.sv
obj_dir//yosyshq/share/verilator/include/verilated_std.sv
obj_dir//yosyshq/share/verilator/include/verilated_std.sv.v
obj_dir//yosyshq/share/verilator/include/verilated_std.sv.sv
%Error: Exiting due to 2 error(s)
scons: *** [hardware] Error 1
====================================================== [ ERROR ] Took 0.11 seconds ======================================================
```

I have tested all the other commands (that does not require hardware) and apio sim/build/time run fine. Did I miss an install step? I can provide the and files, but its pretty basic :)

Obijuan, 2024-07-07:
Hi @obtitus, I can confirm that this is a bug. Thanks for reporting

zapta, 2024-10-13:
This was fixed (not released yet).  The verilator command for ECP5 ``apio lint`` looks now like this:

```
verilator --lint-only --bbox-unsup --timing -Wno-TIMESCALEMOD -Wno-MULTITOP -I"/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ecp5" hardware.vlt main.v main_tb.v
```

and hardware.vlt is an auto generated file that looks like this:

```
`verilator_config
lint_off -rule COMBDLY      -file "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ecp5/*"
lint_off -rule WIDTHEXPAND  -file "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ecp5/*"
lint_off -rule PINMISSING   -file "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ecp5/*"
lint_off -rule ASSIGNIN     -file "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ecp5/*"
lint_off -rule WIDTHTRUNC   -file "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ecp5/*"
lint_off -rule INITIALDLY   -file "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ecp5/*"
```

Closing.

closed 2024-10-13 by zapta