FPGAwars/apio #391: SystemVerilog Support

woolseyj, 2024-08-10:
APIO is a great utility and I use it frequently for testing my Verilog designs.

Are there any plans to support SystemVerilog?

zapta, 2024-10-13:
@woolseyj, apio is base on yosys, nextpnr and the other open sources tools. Do they support System Verilog?

If not, I suggest to file this feature request yes. If yes, please provide informaiton on what apio should do differently (e.g. different flags when invoking yosys or nextpnr).

gmsanchez, 2024-12-10:
@zapta To my understanding, SystemVerilog support can be added by using the `yosys-slang` plugin, which can be obtained in [https://github.com/povik/yosys-slang](https://github.com/povik/yosys-slang). According to `yosys-slang` plugin, it is available prebuilt as part of OSS CAD Suite from YosysHQ.

The OSS CAD Suite that APIO installs does not support that plugin. I have tried the following
```
apio raw yosys
yosys> plugin -i slang
ERROR: Can't load module `./slang': /home/gsanchez/.apio/packages/tools-oss-cad-suite-0.0.9/lib/../share/yosys/plugins/slang.so: cannot open shared object file: No such file or directory
```

 I will try to update the OSS CAD Suite in APIO and see if I can get an example with SystemVerilog to work

zapta, 2024-12-10:
@gmsanchez, the dev version of Apio installs all the files from the yosys oss package but I still get the same error as you.

The yosys hq version we use for apio dev is below.  Can you install it directly from there and see if you can make system verilog to work.  IRRC, it contains a script that you need to run first to set the env variables on your system.

Once we will know how to run SV with the yosys hq package we will know how to replicate it with apio.

https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2024-08-02

zapta, 2024-12-10:
> @zapta <https://github.com/zapta> thanks for the quick response. I was
> able to create an example that works on my EDU-CIAA FPGA
>
> You can check
> https://github.com/gmsanchez/edu-ciaa-fpga-verilog/tree/main/01-leds-sv
>
> In there you can find a file named Info.md that has the list of commands
> I used to get the leds.sv file running on the FPGA.
>
> I downloaded OSS CAD Suite from
> https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2024-12-10,
> which comes with Yosys 0.47+211 (git sha1 b0708a38b, clang++ 18.1.8 -fPIC
> -O3)

Tracks. I will give it a try.

zapta, 2024-12-10:
I was able to reproduce your success with the oss 2024-12-10 version. We
will upgrade the dev oss package to 2024-12-10 and will go from there.

gmsanchez, 2024-12-10:
That's great! Thanks!

I used those commands just because they are based on the workflow I had before using APIO. I know that maybe nowadays nextpnr is preferred over arachne-pnr.

Let me know if you need some help with something.

zapta, 2024-12-11:
@gmsanchez, I sent a PR for the new apio's dev oss-cad-suite package using the latest yosysHQ's package.   Also tried it with a local new package, took apio build commands and use them as are but replacing leds.v instead of leds.sv (this is with the new dev apio which is slightly different from the prod apio). Results below.

A question, with Yosys, can a project has a mix of verilog and system verilog files or do they need to be of same type?  What about the testbenches, can they also be a mix of .s and .sv?  

My shell script test.sh:
```
~/work/edu-ciaa-fpga-verilog-main/01-leds-sv$ cat test.sh
#!/bin/bash

set -e
set -x

apio clean

mkdir _build

apio raw yosys -p "synth_ice40 -top leds -json _build/hardware.json" -q leds.sv

apio raw nextpnr-ice40 --hx8k --package tq144:4k --json _build/hardware.json --asc _build/hardware.asc --report _build/hardware.pnr --pcf leds.pcf -q

apio raw icepack _build/hardware.asc _build/hardware.bin

ls -al _build

```

Execution log:
```
~/work/edu-ciaa-fpga-verilog-main/01-leds-sv$ ./test.sh 
+ apio clean
Setting the envinronment.
Removed _build/hardware.json
Removed _build/hardware.asc
Removed _build/hardware.pnr
Removed _build/hardware.bin
Removed directory _build
=================================================== [SUCCESS] Took 0.06 seconds ===================================================
+ mkdir _build
+ apio raw yosys -p 'synth_ice40 -top leds -json _build/hardware.json' -q leds.sv
Setting the envinronment.
cmd = ['yosys', '-p', 'synth_ice40 -top leds -json _build/hardware.json', '-q', 'leds.sv']
Exit status [0] OK
+ apio raw nextpnr-ice40 --hx8k --package tq144:4k --json _build/hardware.json --asc _build/hardware.asc --report _build/hardware.pnr --pcf leds.pcf -q
Setting the envinronment.
cmd = ['nextpnr-ice40', '--hx8k', '--package', 'tq144:4k', '--json', '_build/hardware.json', '--asc', '_build/hardware.asc', '--report', '_build/hardware.pnr', '--pcf', 'leds.pcf', '-q']
Exit status [0] OK
+ apio raw icepack _build/hardware.asc _build/hardware.bin
Setting the envinronment.
cmd = ['icepack', '_build/hardware.asc', '_build/hardware.bin']
Exit status [0] OK
+ ls -al _build
total 2816
drwxr-xr-x  6 user  staff     192 Dec 10 18:50 .
drwxr-xr-x@ 8 user  staff     256 Dec 10 18:50 ..
-rw-r--r--  1 user  staff  948960 Dec 10 18:50 hardware.asc
-rw-r--r--  1 user  staff  135100 Dec 10 18:50 hardware.bin
-rw-r--r--  1 user  staff  351005 Dec 10 18:50 hardware.json
-rw-r--r--  1 user  staff     312 Dec 10 18:50 hardware.pnr
```

gmsanchez, 2024-12-11:
@zapta awesome work! It seems that everything is working properly. I will try those commands as soon as I have the FPGA with me again.

To my understanding, SystemVerilog is a super set of Verilog and because of that, it should be backwards compatible with Verilog.

In practice, I have never used SystemVerilog until yesterday, so I really don't know the answer. My intuition says that if you use SystemVerilog, then you could use a mix of SystemVerilog and Verilog files. But I am not 100% sure.

zapta, 2024-12-11:
Do you use apio prod or dev?

If you want to use dev it is relatively easy.  Get a copy of the apio
repository (via zip download or git clone) and then

pip uninstall apio    # delete the prod apio
cd <root-directory-of-the-repo>
pip install -e .

To test you can run

apio system -i

gmsanchez, 2024-12-11:
I have always used apio prod and always downloaded it using pip.

I will do that in order to use apio dev and check these features out. Is there anything particular you would like me to test?

gmsanchez, 2024-12-17:
@zapta I have modified some code on `apio/scons/scons_util.py` and as a result of those modifications, now `apio build` works with `.sv` files

You can check https://github.com/gmsanchez/apio/commit/cfe6b315b9f08d0d10efbf40bba92469a54b41b4

woolseyj, 2024-12-17:
@gmsanchez, @zapta, thank you both for working on this.

zapta, 2024-12-17:
Thanks @gmsanchez. that's good news.

I am working now on a large change that touch numerous files so will add this change (with tests, etc) after that.

Can you send a pull request with a test example that includes sv source and testbench files? This will help with the testing. Try to use in the sv file features that are beyond verilog and would fail if used in a .v file.  Extra points if you add two examples, one for the alhambra-ii (ICE40) and one for the ColorLight (ecp5)

https://github.com/FPGAwars/apio/tree/develop/test-examples/TB

gmsanchez, 2024-12-17:
@zapta I have added an example for the EDU-CIAA-FPGA, it is a simple 1-bit AND gate and is available here https://github.com/gmsanchez/apio/commit/e138c5b620746e317ff7f7c54dba0c3c52318e9b

The module uses `logic` when defining the input and output ports instead of `wire`. That change should make it fail if it is used in a .v file

Running `apio build` works, but failed while running `apio sim and1_tb.sv`.

In order to make `apio sim and1_tb.sv` work, I had to change the way `iverilog` is called by adding `-g2012`, you can see that here https://github.com/gmsanchez/apio/commit/ccafb134e56b9cfbbe946b2445fd1ebcf15c07b0

If you think there is a better way to change the way `iverilog` is called instead of hard-coding the parameter that way, let's fix that first. Would that be OK?

zapta, 2024-12-17:
> iverilog is called by adding -g2012, you can see that here gmsanchez@
> ccafb13

Will the -g2012 work also with existing verilog projects or is it something
that we need to control on/off?

gmsanchez, 2024-12-17:
From the `iverilog` man page [https://linux.die.net/man/1/iverilog](https://linux.die.net/man/1/iverilog):

> -g1995|-g2001|-g2001-noconfig|-g2005
Select the Verilog language generation to support in the compiler. This selects between IEEE1364-1995, IEEE1364-2001, or IEEE1364-2005. Normally, Icarus Verilog defaults to the latest known generation of the language. This flag is most useful to restrict the language to a set supported by tools of specific generations, for compatibility with other tools.

So I think that that flag is something that should be turned on/off depending if there are `.sv` files in the project directory. At least for now that approach seems safer and less prone to break stuff that is programmed using only Verilog.

I also created a D_FlipFlop example to test if the `always_ff` primitive works. So far, all good. You can check that here https://github.com/gmsanchez/apio/commit/b21ed1e9c7070a982f3a1d029d852695381b2ea9

zapta, 2024-12-17:
Does this mean that .sv and .v files can't be mixed in the same project?

I am working now on making apio.ini a required file so we can add there
controls but it would be nice if we can mix .v and .sv files  (I am not a
verilog expert so don't know if this is common or useful).

gmsanchez, 2024-12-17:
From what I understand, SystemVerilog is a super-set of Verilog. So, if we use a SystemVerilog enabled toolchain we could mix .sv and .v files.

That being said, I really don't know how mature is the SystemVerilog support in the programs that are being used. For example, if I am not mistaken, SystemVerilog support for the OSS CAD Suite was added on november 15, 2024.

Check this out [https://github.com/YosysHQ/oss-cad-suite-build/commits/main/](https://github.com/YosysHQ/oss-cad-suite-build/commits/main/)

SystemVerilog support was added by adding the `slang` plugin https://github.com/YosysHQ/oss-cad-suite-build/commit/085dd31915ec68bef949d8234cc3c89681586850

Because of this, I think that the most conservative thing would be some of the following:

1. if a project only contains .v files, use the already tested way. Do not enable SystemVerilog support.
2. if a project contains .sv files (which **could** be mixed with .v files), enable SystemVerilog support for all the toolchain.

zapta, 2024-12-17:
Hi @gmsanchez, the two policy items you suggested sound good, safe, and feasible.

gmsanchez, 2024-12-17:
@zapta that is great!

Is there any way I can help with that? Just let me know

zapta, 2024-12-18:
Hi @gmsanchez, I was able to reproduce your success but ``apio graph`` still fails.   Can you take a look?  Attached is my sv proj.  

[and-gate.zip](https://github.com/user-attachments/files/18174759/and-gate.zip)

```
/projects/apio-dev/repo/test-examples/TB/edu-ciaa-fpga/and-gate$ apio graph
Setting the envinronment.
Warning: project contains .sv files, system-verilog support is experimental.
yosys -f verilog -p "show -format dot -colors 1 -prefix _build/hardware and_gate" -q and_gate.sv
and_gate.sv:2: ERROR: syntax error, unexpected TOK_ID, expecting ',' or '=' or ')'
scons: *** [_build/hardware.dot] Error 1
```

gmsanchez, 2024-12-18:
Hi!

If I unpack the zip file and inside that directory I run:

`yosys -m slang -p "read_slang and_gate.sv; show"`

The schematic shows up. I think that the main reason the previous command fails is because of the `-f verilog` parameter, but I can't find what is the yosys-slang frontend name... I will keep working on this

gmsanchez, 2024-12-18:
Hi!

I got it working by running these commands inside the `and-gate` directory

```
apio clean
apio build
yosys -m slang -f slang -p "show -format dot -colors 1 -prefix _build/hardware and_gate" -q and_gate.sv
dot -Tsvg _build/hardware.dot -o _build/hardware.svg
```

Let me know if that works for you

zapta, 2024-12-18:
Hi @gmsanchez, when I use the slang options I get graph failues with a few of the apio examples. For example, the one below ``upduino31/blinky``.

BTW, I think that 'SB_RGBA_DRV' module is defined in the yosys library ``/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v``, and when we run ``apio lint`` we have to specify it explicitly in the verilator command. 

Any suggestions?  Ideally we will have a graph command line that will work for both .v and .sv files. If it's a yosys issue you can ask them directly.

```
/projects/apio-examples/repo/examples/upduino31/blinky$ apio graph
Setting the envinronment.
yosys -m slang -f slang -p "show -format dot -colors 1 -prefix _build/hardware main" -q leds.v main.v oscilator.v
leds.v:37:3: error: unknown module 'SB_RGBA_DRV'
  SB_RGBA_DRV #(
  ^~~~~~~~~~~
ERROR: Compilation failed
scons: *** [_build/hardware.dot] Error 1
```

gmsanchez, 2024-12-18:
I have asked about that issue here: https://github.com/YosysHQ/oss-cad-suite-build/issues/133

In the meantime I will try to figure this out

Thanks!

zapta, 2024-12-18:
> I have asked about that issue here: YosysHQ/oss-cad-suite-build#133
>
> In the meantime I will try to figure this out
>
> Thanks!

Thanks!  It helps a lot.

zapta, 2024-12-18:
FYI,
https://stackoverflow.com/questions/79292059/does-the-systemverilog-standard-allows-mixing-with-verilog-files

gmsanchez, 2024-12-19:
That's what I also think about mixing .sv and .v files.

What I am not sure about is the level of maturity of each of the tools implied in the process. It seems that Verilog is very much supported by the toolchain while SystemVerilog support seems to be a work in progress.

gmsanchez, 2024-12-19:
It seems that the error we are getting is because of the way `slang` plugin handles a `(* blackbox *)` attribute with parameters. It appears that would be the only case, since the `slang` plugin supports blackboxes (without parameters).

You can check

https://github.com/YosysHQ/oss-cad-suite-build/issues/133
https://github.com/povik/yosys-slang/issues/72

It doesn't fail earlier because when we execute `apio build` we are running 

`yosys -m slang -p "synth_ice40 -top main -json _build/hardware.json" -q leds.v main.v oscilator.v`

if you remove the `-q` parameter, you can see that yosys uses a SystemVerilog frontend. But I am not sure what version of SystemVerilog and in what percentage that frontend supports the language. If we force the command to use the `slang` frontend, we can see that fails in the same way the `apio show` command fails

`yosys -m slang -f slang -p "synth_ice40 -top main -json _build/hardware.json" -q leds.v main.v oscilator.v`

zapta, 2024-12-19:
Thanks @gmsanchez.   Do we have any lead for possible solution?

Also, can you create an example that demonstrate this issue so we can use for testing. 

https://github.com/FPGAwars/apio/tree/develop/test-examples/TB/edu-ciaa-fpga

zapta, 2025-01-29:
I just tested the latest yosys version and it still fails on the graph command.

https://github.com/YosysHQ/yosys/issues/4873

gmsanchez, 2025-01-31:
@zapta you are the best!
I will try some examples on these days. Thanks for the hard work

zapta, 2025-01-31:
The changes are still in my private directory. Will send a PR today. Once
accepted, you can install the latest apio dev based on the instructions
here
https://github.com/FPGAwars/apio/wiki/Apio-development-environment#just-trying-the-latest-apio-dev-version

Once you will have the dev apio, you can fetch an SV example to an empty
directory using '

apio examples fetch alhambra-ii/bcd-counter-sv'  and then 'apio build'.

zapta, 2025-03-23:
The system verilog support is implemented and checked in. At this stage it's experimental. Once we gain confidence we will make it official, remove the warning message, and will mention it in the in-app documentation ('apio info' command).

woolseyj, 2025-03-23:
That is excellent news.  Thank you everyone for all of the hard work.

zapta, 2025-03-23:
@woolseyj, this is still in dev and yet to be released. If you want to try it, you can use one of the installer packages that apio now creates daily.  They don't require preinstallation of python and contains all the necessary files.

https://github.com/FPGAwars/apio/actions

zapta, 2025-05-08:
System verilog is now a standard feature of Apio (still in dev branch). Closing.

closed 2025-05-08 by zapta