FPGAwars/apio #392: Signal name in verilog influences time to pad.

jasonsbeer, 2024-08-18:
Using APIO 0.9.5 with Ubuntu 24.04. Having some interesting issues with the behavior of an iCE40HX4K FPGA. **The verilog synthesizes without error in each case described.** The verilog in question is a bus sizing interface handling two two-way data ports for interfacing a 32-bit CPU with 16-bit and 32-bit target devices. With the verilog in its current state, the FPGA works, but making normally innocuous changes, such as renaming a signal, causes the FPGA to fail. The failure mode is seen as a delayed data signal between the two data busses, which misses the necessary clock edge. The two cases presented strike me as very odd and extremely difficult to troubleshoot.

Build code: apio build --top-module U111_TOP --fpga iCE40-HX4K-TQ144

Case 1: Lines 294-297 and 319-341 of U111_TOP.v . I want to rename the four signals D[0..4]_LATCH to RD[0..4]_LATCH. After renaming all the instances, the FPGA fails as described. I don't understand how the name of a signal can cause a signal to be delayed.

https://github.com/jasonsbeer/AmigaPCI/blob/9f09c65e6f00d8f15b5e7cb5fe2c94e7d0d828d5/Logic/Verilog/Local%20Bus%2068040/U111/U111_TOP.v#L319-L341

Case 2: Lines 206-210 of U111_TOP.v are parameters defining the states of a finite state machine. The values themselves should be irrelevant, except that they are unique from each other. When the value of state LONG_TRANSFER is set to 3'b101, the verilog/FPGA works as expected. When LONG_TRANSFER parameter is set to value  3'b100, the code/FPGA fails. Since these parameters only serve to identify the states of the state machine, I don't understand how this can cause some signals to be delayed.

https://github.com/jasonsbeer/AmigaPCI/blob/9f09c65e6f00d8f15b5e7cb5fe2c94e7d0d828d5/Logic/Verilog/Local%20Bus%2068040/U111/U111_TOP.v#L206-L210

Files can be found here: https://github.com/jasonsbeer/AmigaPCI/tree/main/Logic/Verilog/Local%20Bus%2068040/U111

I'm happy to supply any additional information.

zapta, 2024-08-28:
I am not an expert but since you didn't get yet an answer, here is my 2c.

APIO is a user friendly wrapper around yosys and the issue you mentioned seems to be yosys related, so I suggest to file an issue in the yosys repository  https://github.com/YosysHQ/yosys

The yosys people are not necessary familar with APIO so make sure to post your issue in yosys terminology and command lines. You can see those command in the APIO log when you run the build command.

zapta, 2024-10-13:
@jasonsbeer, this issue seems to belong to yosys. Please file it there. Link above.

As a layman, name change may affect the internal ordering of networks and nodes and affect the results of the various optimizers, e.g. resulting with different delay times.  But the yosys people should know better.

Closing this one. If you think it's apio related (e.g. apio needs to issue a different yosys or nextpnr command lines), please post the additional info here.

closed 2024-10-13 by zapta