FPGAwars/apio #408: Some commands are broken in native exe mode.

zapta, 2024-09-05:
To reproduce, add exe-mode=native to apio.ini and run the apio test or apio sim.  Example below.  Notice that the ``/ice40/cells_sim.v`` path doesn't have the prefix to the apio packages. 

Native mode should apply to binaries but not to apio resources.

```
$ apio  test -t  main_tb.v
Loading custom boards.json from project dir
Warning: native exe mode (binaries should be on path)
iverilog -o main_tb.out -D VCD_OUTPUT=main_tb -D NO_ICE40_DEFAULT_ASSIGNMENTS "/ice40/cells_sim.v" ftdi_tx.v i2s_rx.v i2s_test_pattern.v i2s_timing.v lcd_i2c.v lcd_i2c_io.v leds.v main.v queue.v queue_pusher.v register.v reset_gen.v main_tb.v
/ice40/cells_sim.v: No such file or directory
No top level modules, and no -s option.
scons: *** [main_tb.out] Error 1
```

zapta, 2024-09-05:
The problem seem to be the YOSYS_LIB environment variable that is resolved and set in default mode but left unset (empty) in native mode.  

https://github.com/FPGAwars/apio/blob/154592df13bad0cc9409f6cd8ca4bca55810ae12/apio/managers/scons.py#L920

The paths are used in the link below. Possibly same problem applies to other paths.

https://github.com/FPGAwars/apio/blob/154592df13bad0cc9409f6cd8ca4bca55810ae12/apio/resources/ice40/SConstruct#L81

Another approach would be to deprecate the native exe mode for the benefit of code simplicity.

zapta, 2024-11-12:
@Obijuan, @cavearr, the concept of the native exe mode is inherently problematic because the packages include not just binaries but also libraries and data files (e.g. cell libraries).  Finding a binary on the path is simple but finding it's related support files is problematic. It's safer to work with proper and complete packages.

If there is no objects, I will remove the code related to native exe.

Obijuan, 2024-11-12:
yes, no problem. Go ahead

zapta, 2024-11-12:
Will do. Thanks.

zapta, 2024-11-26:
Native mode code removed. Closing.