FPGAwars/apio #410: apio time - No such option: --top-module

jasonsbeer, 2024-09-06:
Referencing the documentation here : https://github.com/FPGAwars/apio/wiki/Apio-Time

I am trying to run the apio time command while defining the top module, but it throws an error. Has this option been deprecated? Running Ubuntu 24.04.

`(.venv) jason@jason-desktop:~/Documents/U111$ apio time --top-module U111_TOP --fpga iCE40-HX4K-TQ144
Usage: apio time [OPTIONS]
Try 'apio time -h' for help.

Error: No such option: --top-module`

zapta, 2024-09-06:
Documentation may need to be updating. The next release will have more detailed and updated help information.

The recommended way to use apio is to rely on apio.ini for configuration rather than using --top-module, --board, --fpga, and so on.  Try to set apio.ini file and then run 'apio time'.  Same for 'apio build', and 'apio upload'.

For example, my apio.ini looks like this
```
[env]
board = upduino31
top-module = main
```

jasonsbeer, 2024-09-06:
That helped. Thanks!

zapta, 2024-09-06:
Thanks for the update. Keeping this issue open, will look into adding to the ``time`` command a  ``--top-module`` flag for completness.

zapta, 2024-10-13:
The apio command has --top-module flag (possibly added since then).  Closing.

```
$ apio time -h
Usage: apio time [OPTIONS]

  The time command analyzes and reports the timing of the design. It is useful
  to determine the maximal clock rate that the FPGA can handle with this
  design. For more detailed timing information, inspect the file
  'hardware.rpt' that the command generates. The commands is typically used in
  the root directory of the project that contains the apio.ini file.

  Examples:
    apio time

  The time command supportw ICE40 devcies. ECP5 and Gowin devices are not
  supported yet.

  [Note] Flags marked with [DEPRECATED] are not recomanded for use. For
  project configuration, use an apio.ini project file and if neaded, project
  specific 'boards.json' and 'fpga.json' definition files.

Options:
  -p, --project-dir path  Set the root directory for the project.
  -v, --verbose           Show the entire output of the command.
  --verbose-yosys         Show the yosys output.
  --verbose-pnr           Show the pnr output.
  -t, --top-module name   [DEPRECATED] Set the top level module name.
  -b, --board str         [DEPRECATED] Set the board.
  --fpga str              [DEPRECATED] Set the FPGA.
  --size str              [DEPRECATED] Set the FPGA type (1k/8k).
  --type str              [DEPRECATED] Set the FPGA type (hx/lp).
  --pack str              [DEPRECATED] Set the FPGA package.
  -h, --help              Show this message and exit.
```

closed 2024-10-13 by zapta