FPGAwars/apio #422: Regression with Alhambra-II  'apio time'.

zapta, 2024-09-23:
@embedded-tek, @Obijuan 

This commit seems to break the 'apio time' command with Alhmabra-II due to the change of the FPGA size from ``8k`` to ``4k``.

https://github.com/FPGAwars/apio/commit/9b528b0e2f3cb6190140a262d9ab9ebaa44f7e3a

Below is a log fo running the dev apio command on the ``test-examples/Alhambra-II/02-jumping-LED example``. The 4k FPGA size constructs ``CHIPDB_PATH`` with the file name ``chipdb-4k.txt`` which does not exist  (the file ``chipdb-u4k.txt`` does exist).

```
/Users/user/projects/apio_dev/repo/test-examples/Alhambra-II/02-jumping-LED $ apio system -i
Platform: darwin_arm64
Package:  /Users/user/projects/apio_dev/repo/apio
/Users/user/projects/apio_dev/repo/test-examples/Alhambra-II/02-jumping-LED $ apio clean
===================================================================== [SUCCESS] Took 0.06 seconds =====================================================================
/Users/user/projects/apio_dev/repo/test-examples/Alhambra-II/02-jumping-LED $ apio time
[Sun Sep 22 22:15:38 2024] Processing alhambra-ii
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
yosys -p "synth_ice40 -top main -json hardware.json" -q main.v
Warning: wire '\q' is assigned in a block at main.v:3171.14-3171.21.
Warning: wire '\q' is assigned in a block at main.v:3175.6-3175.12.
nextpnr-ice40 --hx4k --package tq144 --json hardware.json --asc hardware.asc --pcf main.pcf -q
icetime -d hx4k -P tq144 -C "/Users/user/.apio/packages/tools-oss-cad-suite/share/icebox/chipdb-4k.txt" -mtr hardware.rpt hardware.asc
// Reading input .asc file..
// Reading 8k chipdb file..
Can't open chipdb file: No such file or directory
scons: *** [hardware.rpt] Error 1
===================================================================== [ ERROR ] Took 2.00 seconds =====================================================================
/Users/user/projects/apio_dev/repo/test-examples/Alhambra-II/02-jumping-LED $ 
```

embedded-tek, 2024-09-23:
I haven't used this `apio time` functionality since it never worked for me. Probably because of this same issue. I am not sure what the fix would be since this issue is not an implementation problem with the addition of the new board and/or fixing the `fpgas.json` file to correctly reflect the proper LUT size. 

I would be happy to fix the icetime function, but I am unsure how to generate the new chipdb file.

zapta, 2024-09-23:
I think that the first step is to determine what what chipdb file should be used for Alhambra-II (the canonical FPGAWars board). The candidates are the two 4k chipdb files that are currently installed (see below) or a new one that need to be added. 

Nextpnr seems to have a generator for these files https://github.com/YosysHQ/nextpnr/blob/master/ice40/chipdb.py

@Obijuan, any thoughts?  

```
/Users/user/.apio/packages/tools-oss-cad-suite/share/icebox $ ls -1
chipdb-1k.txt
chipdb-384.txt
chipdb-5k.txt
chipdb-8k.txt
chipdb-lm4k.txt
chipdb-u4k.txt
timings_hx1k.txt
timings_hx8k.txt
timings_lp1k.txt
timings_lp384.txt
timings_lp8k.txt
timings_u4k.txt
timings_up5k.txt
```

embedded-tek, 2024-09-23:
I agree. I read through the files earlier and I am not sure which ICE40 the `u4k` belongs to. I do know it is not compatible with the HX4K because it is a higher pin count versus the sg48 defined in the file. I would think a new file would have to be generated.

zapta, 2024-09-23:
I also wonder if this formula for the chipdb file name is sufficient.  

https://github.com/FPGAwars/apio/blob/88672ad6ccffffabaf65487ea5dbdc906ffad2d0/apio/resources/ice40/SConstruct#L80

```
CHIPDB_PATH = os.path.join(ICEBOX_PATH, 'chipdb-{0}.txt'.format(FPGA_SIZE))
```

1. It ignores the the FPGA 'type' value.
2. It's the same for ice40, gowin, and ecp5.

@Obijuan , can you revisit the chipid generation and runtime selection?

Obijuan, 2024-09-23:
The `u4k'  family does not exist in reality. There are only two diffent manufactured families: 1K and 8K.   The 4K family in an 8K that is actually not fully used by the Lattice software (it is the propietary software that limit you from using the 8K). Let's say that 4K is a 'virtual' FPGA limited by software. When you use the Open source tools, you can see that any 4K FPGA is in fact an 8K. So, the Alhambra-II board es an 8K. Any board with 4K FPGA, is an 8K

zapta, 2024-09-23:
Thanks @Obijuan, this is very useful. A couple of follow up questions for you (referring to the ice40 fpga list below from the dev repository)

1. What does the ':4k' in the line below mean? Can we drop it? Does it limit the yosys design to 4k?
```
iCE40-HX4K-BG121                ice40    hx           8k    bg121:4k
```
2. What about the 1k and 5k, are these real sizes or is the 5k for example is actually 8k?

3. Is it safe to use a 4k as 8k or are they tested and guaranteed just as 4k?

```
$ apio boards --fpga | grep -i ice 
• iCE40-HX1K-CB132                ice40    hx           1k    cb132     
• iCE40-HX1K-TQ144                ice40    hx           1k    tq144     
• iCE40-HX1K-VQ100                ice40    hx           1k    vq100     
• iCE40-HX4K-BG121                ice40    hx           8k    bg121:4k  
• iCE40-HX4K-CB132                ice40    hx           8k    cb132:4k  
• iCE40-HX4K-TQ144                ice40    hx           4k    tq144     
• iCE40-HX8K-BG121                ice40    hx           8k    bg121     
• iCE40-HX8K-CB132                ice40    hx           8k    cb132     
• iCE40-HX8K-CM225                ice40    hx           8k    cm225     
• iCE40-HX8K-CT256                ice40    hx           8k    ct256     
• iCE40-LP1K-CB121                ice40    lp           1k    cb121     
• iCE40-LP1K-CB81                 ice40    lp           1k    cb81      
• iCE40-LP1K-CM121                ice40    lp           1k    cm121     
• iCE40-LP1K-CM36                 ice40    lp           1k    cm36      
• iCE40-LP1K-CM49                 ice40    lp           1k    cm49      
• iCE40-LP1K-CM81                 ice40    lp           1k    cm81      
• iCE40-LP1K-QN84                 ice40    lp           1k    qn84      
• iCE40-LP1K-SWG16TR              ice40    lp           1k    swg16tr   
• iCE40-LP384-CM36                ice40    lp           384   cm36      
• iCE40-LP384-CM49                ice40    lp           384   cm49      
• iCE40-LP384-QN32                ice40    lp           384   qn32      
• iCE40-LP4K-CM121                ice40    lp           8k    cm121:4k  
• iCE40-LP4K-CM225                ice40    lp           8k    cm225:4k  
• iCE40-LP4K-CM81                 ice40    lp           8k    cm81:4k   
• iCE40-LP8K-CM121                ice40    lp           8k    cm121     
• iCE40-LP8K-CM225                ice40    lp           8k    cm225     
• iCE40-LP8K-CM81                 ice40    lp           8k    cm81      
• iCE40-U4K-SG48                  ice40    u            4k    sg48      
• iCE40-U4K-UWG30                 ice40    u            4k    uwg30     
• iCE40-UL1K-CM36A                ice40    ul           1k    cm36a     
• iCE40-UP5K-SG48                 ice40    up           5k    sg48      
• iCE40-UP5K-UWG30                ice40    up           5k    uwg30 
```

embedded-tek, 2024-09-24:
Does a new chipdb file need to be created still? The 8k file has the pinout of the BGA and the u4k file is not technically real. I am not sure how important it is to have the pinout be correct. I would be happy to work on generating a file for the 4k (8k technically thank you @Obijuan for the clarification) for the tq144 package but I would need to be pointed to some documentation.

zapta, 2024-09-24:
@embedded-tek, since this issue is involved, how about restoring the original configuration and then tracking the chipdp thing in a separate issue?

embedded-tek, 2024-09-25:
That works. Let me revert and test locally. Then I can look into it a little more.

zapta, 2024-10-04:
@embedded-tek, I am preparing a commit to restore the original definition. This will make the Alhambra II test examples work again with 'apio time'. 

Alhambra II is the reference borad of apio and possibly the most tested, so chances are these definitions should work also with your board (with the same FPGA).

```
  "iCE40-HX4K-TQ144": {
    "arch": "ice40",
    "type": "hx",
    "size": "8k",
    "pack": "tq144:4k"
  },
```

zapta, 2024-10-13:
Fixed.

closed 2024-10-13 by zapta

zapta, 2025-01-16:
A quick update, with the new fpga.jsonc convetions, the fpga id is decoupled from the fpga part number and the same part number can have multiple instances, for example:

```
  "ice40hx4k-tq144": {
    "part_num": "ICE40HX4K-TQ144",
    "arch": "ice40",
    "size": "4k",
    "type": "hx4k",
    "pack": "tq144"
  },
  // This is a 4k fpga that is oversized to 8k.
  "ice40hx4k-tq144-8k": {                               
    "part_num": "ICE40HX4K-TQ144",
    "arch": "ice40",
    "size": "8k",
    "type": "hx8k",
    "pack": "tq144:4k"
  },
```